patience4711/RPI-APS-inverters #2: unable to open serial_port - ECU_V02

kentbrock, 2022-01-02:
After some "debuging" with the RPI I still got the feedback "unable to open serial_port" when starting the coordinator. 

*****************************************
running script startCoordinator.pl...
calling startCoordinator.cgi argument is 1
command to run = sudo /usr/lib/cgi-bin/ecu/checkCoordinator.cgi
calling checkCoordinator.cgi Content-type:text/html
running checkCoordinator.cgi
wiringPi started
unable to open serial_port
result = 0
broke out at 1
startCoordinator runned
HTTP:1/1 200 OK
*****************************************

My configuration: RPI2 with CC2531 (I know this is not the best configuration - I do not have an Zero currently and the CC2530 is in my productive ESP Setup)

What I´ve tried so far:

Using the CC2530 with CC2530ZNP-with-SBL.hex (flashed via Ti HW), (CC2530 is working with ESP) -> unable to open serial port
Using the CC2531 with CC2531ZNP-with-SBL.hex (flashed via Ti HW) -> unable to open serial port
Using the CC2531 with CC2531ZNP-with-SBL.hex (flashed succesfully via RPI - with your ECU_V02) -> unable to open serial port

Next I will double check if the CC2531 is going to work with the ESP image and in addition to that try to get an Zero from a friend of mine.

Maybe its good to know, that the green LED of the CC2531 turns of after a while during boot up.

Do you have other ideas?

patience4711, 2022-01-03:
I  think that the problem is that your zigbee is plugged in an usb port.  The raspberry maybe decides that that is the serial port.  You can try to solder the vcc and ground on the zigbee and wire that to the rpi header. Otherwise i wouldn't know.

kentbrock, 2022-01-04:
I have started from the scratch today. Used the image ecu_v01 and reflashed the CC2531 with the TI Tool. The CC2531 isn´t plugged inside the USB Port and i have check the wiring twice (also switched DD and DC). Always the same behaviour: "unable to open serial port". 
I have also tried a second CC2531 (I have two of them here) - same behaviour.

I think I have to buy a zero and a CC2530 to have an identical setup like you

kentbrock, 2022-01-06:
I think the puzzle is solved. I used a Raspberry 2 Model B and there is a difference in UART handling from RPI 3 Model on.

I found the hint on this website: https://dasraspberrypi.de/enable-port-serial-raspberry-pi-2/

Translated into english:
"All Raspberry models have at least 2 UARTs, number 0, which is a normal UART, and number 1, which is a mini UART with much more limited and very restrictive capacities.
In the past, UART 0 was used by default. However, since this is no longer being upgraded to version 3, it is now used to manage the Bluetooth chip."

With the RPI Zero and also with an RPI3 there is no Problem with opening serial port. Tomorow I try to pair the inverter with the RPI image.