patience4711/RPI-APS-inverters #3: RPI-ECU_v02

Chen32220, 2022-01-08:
Great new values.

For inverter "DOM.IDX "  What is it?
I entered 895 without knowing

patience4711, 2022-01-10:
If you upgraded to the latest release, you could not have these daily energy values for inverter 0. Unless you forgot to make the relative database empty. These values belong to my inverter. You can get rid of them if you go to the menu databases. Choose inverter 0 and click wipe values for both databases.

The graphs for power etc work fine for all inverters?
If i understand well what you mean to say is that there are no daily energy values / statistics for other than inverter 0.?  At the first day this is normal but if this still is the situation please do the following: Go to the menu databases click inverter 1 and click query from invEnergy and send me the screenshot, so that i can see if this database is working or not. 

As for the DOMidx, this is just an id that is sent in mqtt messages to Domoticz, so that the device can be identified. I'l update that in the help page.

Chen32220, 2022-01-16:
_The graphs for power etc work fine for all inverters?_
no: only for inv0

_If i understand well what you mean to say is that there are no daily energy values / statistics for other than inverter 0_.
Yes

[image]
[image]
[image]
s

patience4711, 2022-01-17:
I am testing with a fake inverter that i gave the number 3 and everything seems to work fine. From your pictures i conclude that the databases work fine, there are values in invEnergy which means that the other database is working also. 
Seems that the chart pages do not reed the database correct. You could do the following test:
- select inverter 1 on the front page and todays date in the charts page.
- open a new page and type  'http://**your_ecu_ip**/chartjs/chartdata_pe.php' in the addressbar and enter
- now you should get something like:<br>
[image]
<br>This is the today's power and energy data for the chosen inverter. Do you get anything like this?

Later this week i will release a new version that has the possibility to install software updates. This means that we only have to write the image to the  sd-card once.  Now furure software upgrades can be installed without losing all our settings and databases.

Chen32220, 2022-01-18:
no doubt: all inverters transmit power and energy.
[image]
For inverter 0 everything is fine: detail, chart, power, energy, p&e, stats
Problem: for all the other, 1 2 3 4 5 6 7 8,
 there is no display "stats" and in "energy" the previous days are not displayed
[image]

patience4711, 2022-01-18:
Oke, i found the bug and corrected it. This evening after sunset i'l release the new improved image.

Chen32220, 2022-01-23:
Everything is fine with version1_1. Congratulation
I look forward to the file for housing

closed 2022-01-24 by patience4711