patience4711/RPI-APS-inverters #10: RPI-ECU test results

patience4711, 2022-01-26:
Post your test results and user experience here

WaiH, 2022-01-26:
@patience4711 
Can you open another discussion only for testing questions and results ?
so this thread can be spared of that ?

Tested the new image, and it's working fine.  the normal method, want to test the wpa_supplicant.conf method later when i've the time

got this pair message think it's paired.
[806000019224 pair message.txt](https://github.com/Koenkk/zigbee2mqtt/files/7943190/806000019224.pair.message.txt)
on the detail message, the name field is blank.
no sun now, need to test tomorrow.

thanks

patience4711, 2022-01-26:
@WaiH 
According to the file it is paired. So tomorrow it should start polling..

WaiH, 2022-01-26:
Hardware:
RPI zero 2W
Kingston DT100G3/64GB DataTraveler 100 G3, USB 3.0, 3.1 Flash Drive, 64 GB
CC2530 + CC2591 Module Zigbee with firmware: CC2530ZNP-cc2591-with-SBL.hex

The previous test with an esp8366 setup was with another CC2530 + CC2591 Module. 
The mistake i think was that i use the CC2530ZNP-with-SBL.hex. that's why it couldn't pair i think . That module is bad now cause i can't flash it again, ./cc_chipid gives me ffff and not ID = a524. with the same wiring to a rpi 2 for flashing as i did with the new one.

I tried to give the rpi-ecu a static adress, but sofar no luck. need to find what i did wrong.

i see that on the ip-config page you got a typo : Tip: **klick** on "help". mix of NL-EN

patience4711, 2022-01-27:
I never used Ztool, for me is the answer you get on a polling request significant.

WaiH, 2022-01-27:
[image]

this is what i see, no name for the inverter

[image]
there is data.

and inverter page says this, no name etc.

[image]

and i do get MQTT messages format 2

only haven't got to put in Home Assistant sensors correctly.
Bericht 4 ontvangen op domoticz/in om 11:10:

{"serial":"806000019224","acv":0.0,"freq":inf,"temp":,"pwr":[0.0,0.0,0.0,0.0],"totalpwr":0.0,"en":[0.8,0.4,0.0,0.5],"totalen":1.7}

QoS: 0 - Retain: false
Bericht 3 ontvangen op domoticz/in om 11:10:

{"serial":"408000195480","acv":0,"freq":0,"temp":,"pwr":[0,0,0,0],"totalpwr":0,"en":[0.8,0.4,0.0,0.0],"totalen":1.2}

QoS: 0 - Retain: false

patience4711, 2022-01-27:
Seems that there is no proper file for the inverter. Did you do the factory reset? I would start with that and add a new converter with  the right properties.

WaiH, 2022-01-27:
will edit the previous post so itś more readable

patience4711, 2022-01-27:
You did the factory reset, than added an inverter, an yc600. You got values, this should not be possible as it was not yet paired. 
When not paired, the inverter cannot be polled becaues the polling script gets the inverterID from the file. 
So when you pair successfully, this script adds the inverterID to the inverterfile.  Because of this the inverterfile gets malformed.

Maybe since you run this from an usbstick, the write permissions are changed during the copy process. So maybe you should start running this from an sdcard to get a comparable situation.

WaiH, 2022-01-27:
inverterpair running with arg0
wiringPi started
opened port serial0
fileToread = /var/www/ecu_data/ecuProperties
json from file = { 'idinvers': "80971B01B3D6", 'id': "D6B3011B9780" }
idinvers from file = 80971B01B3D6
ecushort = B3D6
fileToread = /var/www/ecu_data/inverters/invProperties0
json from file = { 'panels': [ 1, 1, 0, 0 ], 'idx': "VT", 'id': "unpaired", 'name': "VoorTuin", 'type': 0, 'serial': 408000195480 } serial from file = 408000195480
trying to pair inverter0
send cmd 0 2524020FFFFFFFFFFFFFFFFF14FFFF140D0200000F1100408000195480FFFF10FFFF80971B01B3D668
inMessage = FE0164020067FE25448100000D02000014140127000C4703000011408000195480FFFF10FFFF80971B01B3D642200D8A
send cmd 1 1A24020FFFFFFFFFFFFFFFFF14FFFF140C0201000F060040800019548038
inMessage = FE0164020067FE1C448100000101D97A1414004E00C54D03000008FF1A408000195480D97A0EF2FE1A448100000C0200001414014E00635303000006408000195480D97A0D08
split to find invSerial
result after 1st split = D97A0EF2FE1A448100000C0200001414014E00635303000006408000195480D97A0D08
result after next split = D97A0D08
success, got invID: saving: 0x7AD9
fileToread = /var/www/ecu_data/inverters/invProperties0
save command = echo "{ 'panels': [ 1, 1, 0, 0 ], 'idx': "VT", 'id': '0x7AD9', 'name': 'VoorTuin', 'type': 0, 'serial': 408000195480 }" > "/var/www/ecu_data/inverters/invProperties0"
sedCmd = sed -i 's|['\'']|'\"'|g' /var/www/ecu_data/inverters/invProperties0
returning to function sendCommands
proceeding with the next cmd
send cmd 2 2524020FFFFFFFFFFFFFFFFF14FFFF140F0102000F1100408000195480B3D610FFFF80971B01B3D60E
decode inMessage = FE0164020067FE25448100000F0100001414012700815403000011408000195480B3D610FFFF80971B01B3D642200D70
no pairing answer, returning...
proceeding with the next cmd
send cmd 3 1A24020FFFFFFFFFFFFFFFFF14FFFF14010103000F060080971B01B3D651
decode inMessage = FE0164020067
all pair cmds done
info = 0
successfully paired
pair command processed

patience4711, 2022-01-27:
That looks allright, so in this situation the inverterpage should have the same information in the fields. If not try to refresh the page.

WaiH, 2022-01-27:
will do that later, doing the steps now with SD card

patience4711, 2022-01-27:
That seems oke too. So now the situation is that you have a proper invProperties, the pairing and polling succeeded but the inverterpage shows wrong info?

WaiH, 2022-01-27:
> That seems oke too. So now the situation is that you have a proper invProperties, the pairing and polling succeeded but the inverterpage shows wrong info?

yes,

patience4711, 2022-01-27:
I can reproduce it with your invProperties0, very strange. Investigating ...

WaiH, 2022-01-27:
oke this is what i did,

i edited the invProperties0

i was:
{
"panels": [
1,
1,
0,
0
],
"idx": VT,
"id": "0x7AD9",
"name": "VoorTuin",
"type": 0,
"serial": 408000195480
}

then edit the converter 0 in web page with the same credentials
then the invProperties0 looks like this:

{"id":"unpaired","serial":408000195480,"type":1,"name":"VoorTuin","idx":"VT","panels":[1, 1, 0, 0]}

then i edit it to:

{"id":"0x7AD9","serial":408000195480,"type":0,"name":"VoorTuin","idx":"VT","panels":[1, 1, 0, 0]}

and it looks fine
 did the same with QS1
ok

patience4711, 2022-01-27:
It was the VT, you tweaked that with quotes so now php can read it. If you just provide some integer value it will work too. 
I cannot prevent that field from entering non numeric values so maybe i should handle it as a string. Do that in the next software update.

WaiH, 2022-01-27:
> It was the VT, you tweaked that with quotes so now php can read it. If you just provide some integer value it will work too. I cannot prevent that field from entering non numeric values so maybe i should handle it as a string. Do that in the next software update.

you mean the DOM. IDX ? well yesterday  that filed was empty, i just added that today

WaiH, 2022-01-27:
> The official ECU works with intervalls of 5 minutes and the experiences with the esp-ecu have shown that when the polling rate is higher, stability problems arrise. If you want that you can tweak the cron.
> 
> It turns out that we have found a bug after all. The domidx is initial saved as a string so "123" or "VT". The paircommand adds the inverter id and after that the doublequotes are gone for some reason. So we have 123 and VT then. And there it goes wrong if the value is non-numeric.

Glad i could help, thank you for your patience.

closed 2022-01-28 by patience4711