patience4711/RPI-APS-inverters #14: MQTT messages : "Unexpected token , in JSON at position 55"

WaiH, 2022-01-28:
In node -red  i've got issue with  the payload. 

"{"serial":"806000019224","acv":0.0,"freq":inf,"tem":,"pw0":0.6,"pw1":0.5,"pw2":0.0,"pw3":0.6,"totalpwr":1.7,"en0":0.2,"en1":0.1,"en2":0.0,"en3":0.2,"totalen":0.5}"

"freq":inf  seems if there is a not numeric value it should be in Quote

"{"serial":"408000195480","acv":220.1,"freq":50.0,"tem":,"pw0":1.1,"pw1":1.2,"pw2":0.0,"pw3":0.0,"totalpwr":2.3,"en0":0.4,"en1":0.6,"en2":0.0,"en3":0.0,"totalen":1.0}"

"tem":,  seems empty value is not permitted

It this fixable ?

patience4711, 2022-01-28:
No doubt. 
Seems we are talking about format 3.
The tem is a spell error, thats why it is always n/e.  
The freq value inf  shouldn't be possible, i have neve seen it, when does that occur?

WaiH, 2022-01-28:
freq; inf = infinite, occurs on the QS1, haven´t got to troubleshoot that yet, 
Itś due to that those 4 panels doesn´t give W output . dunno why, i assembled it just like the YC600.
Haven´t got time and good weather to go back on the "outbouw" to rearrange the wiring of the panels to the inverter.

patience4711, 2022-01-28:
Yes. Can you give me a result of a poll command for that inverter? (via the console)

WaiH, 2022-01-28:
running inverterPoll.cgi
inverterpoll running with arg 1
invSerial from file = 806000019224
invType = 1
idx = 224
invIdrev = 4220
idShort = xxxx
idinvers from file = 80971B01B3D6
poll cmd = 1D240142201414060001000F1380971B01B3D6FBFB06BB000000000000C1FEFE55
answer = FE0164010064FE034480001401D2FE0345C4422000E0FE7244810000060142201414005500F54E3100005E806000019224FBFB51B103D00000000000D07600307700E0760190760000423000000002A500023E3D6C00015F9453000011F3AE00018A24701F0003000100000000000000000000000000000000000000000000000000001A7F0000FEFE42200E01
messageToDecode = FE0164010064FE034480001401D2FE0345C4422000E0FE7244810000060142201414005500F54E3100005E806000019224FBFB51B103D00000000000D07600307700E0760190760000423000000002A500023E3D6C00015F9453000011F3AE00018A24701F0003000100000000000000000000000000000000000000000000000000001A7F0000FEFE42200E01
message split after 44810000 060142201414005500F54E3100005E806000019224FBFB51B103D00000000000D07600307700E0760190760000423000000002A500023E3D6C00015F9453000011F3AE00018A24701F0003000100000000000000000000000000000000000000000000000000001A7F0000FEFE42200E01
processing voltage & current
json from file = { 'heath': 9.6, 'dcv': [ 38.2, 38.3, 38.4, 38.2 ], 'freq': inf, 'totalen': 3.1, 'acv': 0.0, 'en': [ 1.3, 0.8, 0.0, 0.9 ], 'dcc': [ 0.0, 0.0, 0.0, 0.0 ], 'idx': 224, 'totalpw': 0.9, 'pwr': [ 0.3, 0.1, 0.5, 0.1 ], 'time': 16915 }
time from file = 16915
energy_old_total from file = 3.1
invType = 1
time from inverter 16944
the timespan = 29
* * * * POLLED INVERTER 1 * * * *
** values panel 0**
extracted energy_ext[x] = 574.000000
energy_new[x] = 1.325143 energy_old[x] = 1.300000
panel 0 power 3.12119 energy today 1.32514
** values panel 1**
extracted energy_ext[x] = 351.000000
energy_new[x] = 0.810322 energy_old[x] = 0.800000
panel 1 power 1.28141 energy today 0.810322
** values panel 2**
extracted energy_ext[x] = 17.000000
energy_new[x] = 0.039246 energy_old[x] = 0.000000
panel 2 power 4.87197 energy today 0.0392464
** values panel 3**
extracted energy_ext[x] = 394.000000
energy_new[x] = 0.909593 energy_old[x] = 0.900000
panel 3 power 1.19083 energy today 0.909593
total inv: power: 10.4654 energy: 3.0843
procentual increase = -0.506309
data lower than former poll, return false
all cmds sent
returned by sendCommands() 1
toLog =
12:55:34 polled inv 1 error 1 energy lower than former poll
logCmd = echo "
12:55:34 polled inv 1 error 1 energy lower than former poll" >> /ramdisk/ecu_log.txt
Content-type:text/html

WaiH, 2022-01-28:
inverterpoll running with arg 1
invSerial from file = 806000019224
invType = 1
idx = 224
invIdrev = 4220
idShort = xxxx
idinvers from file = 80971B01B3D6
poll cmd = 1D240142201414060001000F1380971B01B3D6FBFB06BB000000000000C1FEFE55
answer = FE0164010064FE034480001401D2FE0345C4422000E0FE7244810000060142201414005500962D7E00005E806000019224FBFB51B103CF0000000000207600907600307601E0750000487C00000002A500026D05EA0001822302000011F3E80001B1BBB11F0003000100000000000000000000000000000000000000000000000000001A7F0000FEFE42200E10
messageToDecode = FE0164010064FE034480001401D2FE0345C4422000E0FE7244810000060142201414005500962D7E00005E806000019224FBFB51B103CF0000000000207600907600307601E0750000487C00000002A500026D05EA0001822302000011F3E80001B1BBB11F0003000100000000000000000000000000000000000000000000000000001A7F0000FEFE42200E10
message split after 44810000 060142201414005500962D7E00005E806000019224FBFB51B103CF0000000000207600907600307601E0750000487C00000002A500026D05EA0001822302000011F3E80001B1BBB11F0003000100000000000000000000000000000000000000000000000000001A7F0000FEFE42200E10
processing voltage & current
json from file = { 'heath': 9.6, 'dcv': [ 38.0, 38.1, 38.2, 38.1 ], 'freq': inf, 'totalen': 3.3, 'acv': 0.0, 'en': [ 1.4, 0.9, 0.0, 1.0 ], 'dcc': [ 0.0, 0.0, 0.0, 0.0 ], 'idx': 224, 'totalpw': 0.7, 'pwr': [ 0.3, 0.0, 0.5, 0.0 ], 'time': 18415 }
time from file = 18415
energy_old_total from file = 3.3
invType = 1
time from inverter 18556
the timespan = 141
* * * * POLLED INVERTER 1 * * * *
** values panel 0**
extracted energy_ext[x] = 621.000000
energy_new[x] = 1.433648 energy_old[x] = 1.400000
panel 0 power 0.859086 energy today 1.43365
** values panel 1**
extracted energy_ext[x] = 386.000000
energy_new[x] = 0.891124 energy_old[x] = 0.900000
panel 1 power 0 energy today 0.891124
** values panel 2**
extracted energy_ext[x] = 17.000000
energy_new[x] = 0.039246 energy_old[x] = 0.000000
panel 2 power 1.00204 energy today 0.0392464
** values panel 3**
extracted energy_ext[x] = 433.000000
energy_new[x] = 0.999629 energy_old[x] = 1.000000
panel 3 power 0 energy today 0.999629
total inv: power: 1.86112 energy: 3.36365
procentual increase = 1.92868
echo "{ 'totalpw': 1.9, 'en': [ 1.4, 0.9, 0.0, 1.0 ], 'dcc': [ 0.0, 0.0, 0.0, 0.0 ], 'pwr': [ 0.9, 0.0, 1.0, 0.0 ], 'time': 18556, 'idx': 224, 'acv': 0.0, 'totalen': 3.4, 'freq': inf, 'dcv': [ 38.0, 38.1, 38.2, 38.1 ], 'heath': 9.6 }" > "/ramdisk/invData1"
sedCmd = sed -i 's|['\'']|'\"'|g' /ramdisk/invData1
dbCmd = curl -i -XPOST 'http://localhost:8086/write?db=invData' --data-binary 'inv1 p=1.9,e=3.4'
HTTP/1.1 204 No Content Content-Type: application/json Request-Id: f47699e0-8034-11ec-8052-e45f017adbae X-Influxdb-Build: OSS X-Influxdb-Version: 1.8.10 X-Request-Id: f47699e0-8034-11ec-8052-e45f017adbae Date: Fri, 28 Jan 2022 12:22:26 GMT all cmds sent
returned by sendCommands() 0
inverterpoll exit with code 0
toLog =
13:22:26 polled inv 1 timespan: 141 power: 1.9
logCmd = echo "
13:22:26 polled inv 1 timespan: 141 power: 1.9" >> /ramdisk/ecu_log.txt
command processed
HTTP:1/1 200 OK

patience4711, 2022-01-28:
Oke looks good. Now what exactly is the problem, doesn't the data show up at the frontpage? That could be due to the inf. I am testing your poll answer to see if i can find out why this inf arrises. (and why not with an yc600,  while in the esp the same calculations make no difference).

WaiH, 2022-01-28:
the problem is a correct mqtt message, i am using it in Home Assistant with node Red, and when the message got parsed to a json , it gives an error.

patience4711, 2022-01-28:
Here is a software upgrade to version 1_3<br>
https://1drv.ms/u/s!AkMG5FvUwhediwnN-AgVZRNukZsS?e=lUksFf
If you rename it to sw_update.tar you can upload it and install. (system->update)

The troubles should be over then.

Please let me know how it went.

WaiH, 2022-01-28:
yes thank you ! it works.

patience4711, 2022-01-28:
@WaiH 
I noticed that in your polling answer from the qs1 the values of freq and avc are zero, this is what cause the problems in the first place. 
Cannot understand what the reason is, new inverter version or so? Anyway i added a workaround for this like i did with the 'inf' problem. If the values are inf resp 0 they become 50 resp 220. Untill i find out what is going on here this will do, these values are 50 and 220 anyway.

So what you need to do is upgrade again to version 1_3 with the file you'l find in the chapter 'downloads' . This contains the latest adjustments.

WaiH, 2022-01-28:
The Value from QS1 are low has nothing to do with your program.
It' a fault of my installation of the panels. Haven't found the problem why, cause like i said. 
2 panels with YC600, installed no problem. Then installed 4 panels with the QS1 and checked with Ztool, give me the same low numbers.

Plugged the QS1 on the 2 panels (replace the YC600) and the QS1 gave good numbers.
Plugged the YC600 on 2 of the 4 panels, the same low number.
So i ruled out inverter problems, but since the weather in Amsterdam is not good to troubleshoot, waiting for more sun and at least above 10 degrees to check the panels and cables with multi meter.

patience4711, 2022-01-28:
Oke but you still have a not up-to-date version if you do not upgrade again.

WaiH, 2022-01-28:
updated with the latest update file,

closed 2022-01-28 by patience4711