patience4711/RPI-APS-inverters #15: MQTT Payload usage and trigger timer to request the apsystems

WumpE78, 2022-01-28:
Patience,

where stored the config-file for mqtt-message payload? (Format)

and where can i configure the time to request apsystems-info (5min, is to slow)!!!! with my esp, i trigger every 15sek. (solution, disable database inserts by mqtt send. in my case, data send to dedicated mqtt broker and smart home enviorement (fhem) storage the data to mariadb!

thx and regards
Stefan

patience4711, 2022-01-28:
Hallo Stefan,
/usr/lib/cgi-bin/ecu/sendMqtt.cpp but you should install version1_3 from the sd-card image first, to have the up-to-date .cpp source files. 
I don't understand why 5 minutes (standard aps interval) would be too slow and why some people want to poll preferably every microsecond.  It won't give you better information, only more.  While you look maybe once or twice a day, you would only have to poll once or twice a day.  The polling costs energy as well so you do not improve the yield of your installation. 
But oke,  if you must, you can edit the cron for autopoll.pl. 
viel erfolg ...

WumpE78, 2022-01-28:
`I don't understand why 5 minutes (standard aps interval) would be too slow and why some people want to poll preferably every microsecond. It won't give you better information, only more. While you look maybe once or twice a day, you would only have to poll once or twice a day. The polling costs energy as well so you do not improve the yield of your installation.
But oke, if you must, you can edit the cron for autopoll.pl.`

short request time is only for debugging. I have 90 solarmodules in my garage. these modules should be scrapped. I have to test them before I give them to friends, direct on apsystem when the sun is shining.

thx for support

regards
Stefan

patience4711, 2022-01-29:
In case you missed it: You can easely poll by hand (console), two consecutive tests will give a good picture.

WaiH, 2022-01-29:
i use other interval to sync with SFBspot interval. So they change at the same moment in Home Assistant.

closed 2022-01-29 by patience4711