patience4711/RPI-APS-inverters #16: MQTT Inverter Values are not sent - firmware version = RPI-ECU-v1_2

kentbrock, 2022-01-29:
Hi, 

the test MQTT message works (only) on port 1883. When I change the port to 1884, push the save button and send the test message again it will be send to 1883 still. 

For test purposes I leave the port to 1883 but there are no value messages from the inverter to my MQTT Adapter on the ioBroker.

Why should I use a diffrent port at all? Because I ´am using the Sonoff adapter with the ioBroker and all my sonoff devices are sending the MQTT messages to 1883 an the adapter parses everything. With the NodeMCU implementation everthing works fine with a change of MQTT port.

patience4711, 2022-01-29:
You can ssh to the raspi? If you give me a mailaddress i'l mail a file that you can replace on the raspberry. Then it will work.

kentbrock, 2022-01-29:
Hi,

yes ssh is no problem. Pleas send the file to ***@***.***

Best regards

patience4711, 2022-01-29:
Cannot read it  ***@***.***

kentbrock, 2022-01-29:
mh.. is there a possibilty for a PN via github?

patience4711, 2022-01-29:
download this https://1drv.ms/u/s!AkMG5FvUwhediwwvV2MTELF-7pCG?e=mO2Opm replace the files in /usr/lib/cgi-bin/ecu with these 2.
Check the permissions before and after.

kentbrock, 2022-01-29:
no change. I checked it twice and also the permissions.

patience4711, 2022-01-29:
if you do a ./sendMqtt.cgi you should see what the command is, your port should be in it.

kentbrock, 2022-01-29:
Ah... the Wiring Pi problem again with the "old" RPI2. You must have solved this already, because everything else is working. 

Content-type:text/html

<html>
<head>
<title>RPI ECU PROGAM</title>
</head>
<body>
<h4>running sendMqtt.cgi</h4>
sendMqtt.cgi running with arg 0<br>
wiringPi started
<br>
unable to open serial_port
<br>

patience4712, 2022-01-30:
i am not at home right now but you should be able to repair it yourself. If you open /var/www/cgi-bin/ecu/sendMqtt.cpp  with nano you can find about at the end of the file the open serial routine. The name should be 'serial0'  i think it is now 'ttys0'. 
If you change it to serial0 that you have to compile .
Open 'compile', this file contains compile commands for different files. If there is a command for sendMqtt you can copy that and paste it on the commandline. Otherwise you have to type it yourself in the commandline. 
 
good luck, tomorrow i'l be home.

kentbrock, 2022-01-30:
Perfect - this works for me :-)

Thank you for your great support. 

P.S.: the correct path was /usr/lib/cgi-bin/ecu/sendMqtt.cpp

patience4712, 2022-01-30:
Good, the funny thing is that that program doesn't even use the serial port. That is why i forgot to adapt it.

One remark though, i got the impression that you use version 1_2. If you upgrade to 1_3 (you really should do  that) you have to realize that you have to do this excercise again. Look at the readme under "downloads"

closed 2022-01-31 by patience4711