patience4711/RPI-APS-inverters #18: Does the ESP-ECU also work with a CC2531 USB stick?

jsm78, 2022-02-05:
I use an RPi Zero W 1.1 and a YC600. Your software is already installed. I still have a CC2531 USB stick with the right firmware here, which I like to use. Is there a possibility?

babajun12, 2022-02-05:
...same here. Using a Raspberry Pi3 with CC2531 USB stick. Any chance to SSH into RPi and/or edit device to /dev/ttyACM0 ?
* feature request *

patience4711, 2022-02-07:
Won't work with the stick plugged into the usb. You can ssh (see readme) and try to change the port. In all the programs the portalias serial0 is used. (this i pointing to ttyAMA0).  I think you have to point that to your usb somehow.
I recommend to buy the cc2530, wont cost you much.
Good luck

closed 2022-02-11 by patience4711

babajun12, 2022-02-11:
Update:
Yes and no: You can connect a CC2531 to the PI but not on USB. Just use the serial interface, no USB plugged.
So if you have a CC2531 lying around, you can use it. If you have to buy a zigbee board, go for the CC2530 with a proper antenna.
Health check looks good, unfortunately because of the missing hardware, I connot test the connection to a YC600 - still not sure about purchasing a YC600 or the new DS3.

[image]

`running script checkCoordinator.pl...
calling checkCoordinator.cgi command to run = sudo /usr/lib/cgi-bin/ecu/checkCoordinator.cgi
Content-type:text/html
running checkCoordinator.cgi
wiringPi started
opened port serial0
command = 00270027
answer = FE0E670000FFFF80971B01B3D600000709000F rc=19
fileToread = /var/www/ecu_data/ecuProperties
idinvers from file = 80971B01B3D6
coordinator is running ok
toLog =
21:40:15 checking zb system running ok
logCmd = echo "
21:40:15 checking zb system running ok" >> /ramdisk/ecu_log.txt
command runned
HTTP:1/1 200 OK`