patience4711/RPI-APS-inverters #19: Does it work with the DS3?

daice2k, 2022-02-11:
I was wondering if it works with the DS3, which is the successor of the YC600?

patience4711, 2022-02-11:
Yes we all wonder about that. Someone succeeded in pairing them but got no polling answers. More testing is needed.

daice2k, 2022-02-14:
> Yes we all wonder about that. Someone succeeded in pairing them but got no polling answers. More testing is needed.

I ordered a DS3 and will get it today. I´ll try out if it works.

patience4711, 2022-02-14:
@daice2k  @jenswalit 
To improve our software we heighly depent on guys like you who want to help. Thanks in advance and i am curious about the results. I think that when we can pair it, we also should be able to read it. 
So what i would like to know is that when we trigger a poll, do we get some answer from the inverter. With my software we have a lot of debug options at hand, in the console we can exactly follow the zigbee communication. 

You probably live too far away from my place (Eindhoven) otherwise i could drop by with my laptop to make the sniffs, you probably wouldn't even know haha.

jenswalit, 2022-02-16:
Yes cool, Eindhoven is 200 km away.
My ECU hasn't arrived yet, so I'll let you know what's coming. Then I will try to send you the data from the DS3

daice2k, 2022-02-19:
@patience4711 
> I also would like to propose that we continue this converstation to [Koenkk/zigbee2mqtt#4221](https://github.com/Koenkk/zigbee2mqtt/issues/4221) as the people who inverse engineered the yc600 are very interested in this subject.

OK I have the 2.0 running and try to give the DS3 a bit of DC through the power supply (it´s night here atm) and see what I get and post it at https://github.com/Koenkk/zigbee2mqtt/issues/4221

closed 2022-03-05 by patience4711