patience4711/RPI-APS-inverters #22: Not able to login. Just get to ACCESS FORBIDDEN ! !

vegard1977, 2022-02-17:
Just installed latest version.
Im not able to login in with credentials, i just get to"ACCESS FORBIDDEN ! !"page when i go into "menu"
Im connected to same wifi with laptop and RPI-ECU

patience4711, 2022-02-17:
This is a strange problem, it seems to think that you are form outside.  I tried to reproduce this but it always works in my network. Do you have a wifi repeater in use? Do you never see a login popup from the browser? I use chrome.  Can you try it on a smartphone? What if you type f.i. **http://ip_of_ecu/protected/mosquitto.php**, do you get the login screen?

vegard1977, 2022-02-17:
I have two wifi repeaters.
i tried to turn them off, but same problem presists. ( i will leave them off while i keep testing)
I have never seen the login screen.
I have tried chrome and firefox from my pc.
And chrome from my smartphone.
Same symptoms on all of them.

Tried the direct link, but it goes straight to "acess forbidden" page.

I have no problem logging into rpi.ecu via ssh and can ping outside world from it.

My wifi is a dual 2.4/5 ghz
Can that be a problem ?
I tried first to add it to a 2.4ghz only network, but then the repeaters were on. (same problem then)

Is there any point in trying to connect rpi.ecu to a wired network ?

patience4711, 2022-02-17:
You won't get the login screen when you go to the ecu from outside. So apparently it thinks that you are outside the network, i think that must have something to do with the repeaters. As a workaround you could ssh into the ecu and
edit /etc/apache2/sites-enabled/000-default.conf such that it looks like so:
       <Directory "/var/www/html/protected/">
       # only when logged in from inside lan
          AuthType Basic
          AuthName "Authentication Required"
          AuthUserFile "/etc/htpasswd/.htpasswd"
          <RequireAll>
             <RequireAny>
                Require valid-user
             </RequireAny>
       #         <RequireAny>
       #            Require host localhost
       #            Require ip 127.0.0.1
       #            Require ip 192.168
       #         </RequireAny>
            </RequireAll>
#ErrorDocument 401 /verboden.html
        </Directory>
Than stop and start apache2 (service apache2 stop service apache2 stop)

connect it to a wired network.. you can try that also

vegard1977, 2022-02-17:
Hi

That pointed me in the right direction.
I added Require ip 10.0 instead of using # to comment out.
Now it works just fine, im able to log in  :)

My private lan uses 10.0.0.xxx as default

Thanks so much for your help.
Im really looking forward to connecting my inverter,  im in prosess of mounting it now over the next weeks

vegard1977, 2022-02-17:
Also added " Require ip 10.0" in section above to be able to expand filesystem :)

patience4711, 2022-02-17:
:+1:  :+1:   :+1:

closed 2022-04-30 by patience4711