patience4711/RPI-APS-inverters #24: Implausible values on the day of the update to V2_0

kentbrock, 2022-02-20:
I´ve done the update to V2_0 on the evening of the 18th Februar.  I´ve followed your instructions strictly! Started the backup an hour after sunset and recovered the backup long before midnight.

Everything seems to be fine. Today I´ve seen that there is an implausible value in the "DAILY ENERGY INVERTER 0". The value is far to high. This value leads to high values in the Inverter Statistics. 

Can I manipulate the data point myself so that the calculation fits again? 

[image]

kentbrock, 2022-02-20:
I have noticed that there is a switch in displaying the amount of power over the day:

V 2_0:
[image]

V 1_5:
[image]

These values are realistic. It seems to me that there is switch in the metric of add up the power from 1_5 to 2_0.

patience4711, 2022-02-21:
Hi, The switch in the charts is due to the new approach as to what values are kept in the database. Before the total energy value was shown, now it is the energy increase. There it went wrong with the transistion. It should be done in the time after midnight insteat of before. 
But we can repair that, i'l come back with instructions for that.

closed 2022-03-03 by patience4711

kentbrock, 2022-03-03:
works perfectly! Thank you :-)

patience4711, 2022-05-06:
With the recent versions it is possible to replace or remove wrong values from the energy database via the webinterface.