patience4711/RPI-APS-inverters #25: offset sun-set/rise

Watt600, 2022-02-20:
Thanks for the great software!

I have configured an offset of -30 minutes under the settings:
offset sun-set/rise -30

The info page then shows the following:
sunrise is at : 07:28 and sunset is at 17:43
polling starts at: 06:58 and stops at 17:13

The polling starts half an hour earlier but also ends half an hour earlier.
Shouldn't the polling end better at 18:13?

patience4711, 2022-02-21:
Hi, Yes you are right, I will repair that in the next sw update. For now it is better to start too late with the polling than stopping too early. If you start too late you won't lose data. I think +10 will do.

Watt600, 2022-02-25:
Thank you very much. It works perfectly now!

closed 2022-02-25 by Watt600