patience4711/RPI-APS-inverters #27: cc25 firmware

mamaretti, 2022-02-25:
Hi patience4711

can you please share the firmware for the cc25xx module?

can't find the post of the correct firmware
https://github.com/Koenkk/zigbee2mqtt/issues/4221

i want to use a ByteE18 MS1PA2 but its not working with RPI or ESP so far.

is there any way to check the version of Zigbee module via ESP-Console?

would be great if you can help me.

can u share your binary of ECU right here in Github? 

why you use 1drv.ms?

thx

iLeopod, 2022-02-25:
https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.0.x/bin

patience4711, 2022-02-25:
You should note that the used firmware here is dedicated to this project, developped by @kadszol. 
It is included in this package:
https://github.com/Koenkk/zigbee2mqtt/files/6797510/discord-11-7-2021.zip

mamaretti, 2022-02-26:
thx for the link!

mamaretti, 2022-02-28:
its a distance problem 
E18 MS1PA2 is working when I go closer to inverter

closed 2022-03-03 by patience4711