patience4711/RPI-APS-inverters #28: New image RPI-ECU-v2_0 no start radioAP

b31k, 2022-02-25:
hello good evening, I was trying the new version but it doesn't seem to start the wifi(radioAP), just cloned the image.

patience4711, 2022-02-26:
Sorry, i cannot reproduce this. I just burned the sdcard and run it on a old rpi1, no problem. 
Maybe it is because you have something plugged into the usb? You should only power it via the otg and than (this can take some time) the ap will show up. Sure you have a good class10 sd?

If it doesn't it must be a hardware issue. That cannot find wlan0 is a bad sign. 

Since you have a console you can investigate yourself. You can try raspi-config to connect the wifi but that wouldn't work probably. Or else google is your best friend.
 
You could try to run a official raspberry immage to see if the wifi is working.

Good luck.

b31k, 2022-02-26:
Yes, I just burned the image to an sd card and turned on without any peripheral, and waited more than 30 min.
in fact the RPI-ECU-v1-4 version works perfectly under the same conditions.

patience4711, 2022-02-26:
Oke, thats strange because in the field of wifi these versions are completely equal.  It should not take 30 minutes, maybe 3 at most. 
Wel, if it was the software, others would have the same problem but that seems not the case. 

If the AP doesn't show, maybe it is connected already somehow. 
So the hardware must be oke, than the only thing that could be the cause is the sdcard.  I am trying to download the package and install that image but that doesn't work this moment (network problems). But if that doesn't yield anything i really wouldn't know how i can help.

b31k, 2022-02-27:
sorry no device is listed wlan X

patience4711, 2022-03-01:
So did you try to get your board working with another image?

b31k, 2022-03-01:
Yes, now running ok!
thank very much