patience4711/RPI-APS-inverters #29: RPI-ECU-v2_1 without WiFi?

ToxSox, 2022-03-13:
In the documentation it was written that the software also runs on a "goodold raspi 1".
I have a Raspberry Pi 1B and would work with it right now. However, the device has no WiFi and I would connect it via Ethernet.
Will this work or will the system hang because of the lack of WiFi?

ToxSox, 2022-03-15:
The system starts but doesnt get an ip adress on eth0.

patience4711, 2022-03-16:
@ToxSox 
Just checked and this works:
If you can ssh into the raspberry, add this to /etc/network/interfaces right after iface lo inet loopback:

**auto eth0
allow-hotplug eth0
iface eth0 inet dhcp**  

Disable the service wificheck
#systemctl disable wificheck

#reboot

closed 2022-03-29 by patience4711

rasor89, 2022-05-25:
@patience4711 How can I make an SSH connection with raspberry without IP?