patience4711/RPI-APS-inverters #31: inverters do not communicate well in cloudy days

b31k, 2022-03-22:
Good afternoon
I have a little problem when I have a cloudy day, for example clouds appear and are removed, the inverters do not communicate well.
However, if it is a sunny day from early in the morning it works fine all the time.

EVENTS JOURNAL

2022-03-22 01:00:00 running midNight.cgi
00:00:03 midNight.cgi moved energy total into db invEnergy
midNight: removed data files
sun.py : calculated sunrise / sunset
00:04:03 checking zb system failed, need start (6700) command
00:04:17 checking zb system running ok
07:00:06 woke up from sleep, goodmorning!
going to perform the morning ritual
invdataNull.cgi: wrote a new /ramdisk/invData0 with null values
invdataNull.cgi: wrote a new /ramdisk/invData1 with null values
invdataNull.cgi: wrote a new /ramdisk/invData2 with null values
Wrote 0 in database invData,
1st poll in 5 min, now exit
07:05:07 polled inv 0 error 10 AF_DATA_REQ fail
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData0
07:05:08 polled inv 1 error 10 AF_DATA_REQ fail
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData1
07:05:10 polled inv 2 error 10 AF_DATA_REQ fail
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData2
07:10:07 polled inv 0 error 10 AF_DATA_REQ fail
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData0
07:10:09 polled inv 1 error 10 AF_DATA_REQ fail
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData1
07:10:10 polled inv 2 error 10 AF_DATA_REQ fail
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData2
07:15:07 polled inv 0 error 10 AF_DATA_REQ fail
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData0
07:15:08 polled inv 1 error 10 AF_DATA_REQ fail
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData1

....
....
[image]
[image]

patience4711, 2022-03-28:
@b31k Sorry for my delayed reaction, too busy with other problems. Well i cannot recognize this problem, my raspberry has runned for 38 days without missing one single poll. So i couldn't say what could be the cause of this. From your chart i conclude that the communication failed during the whole day. Never seen that. What zigbee module?

closed 2022-03-31 by patience4711

b31k, 2022-06-12:
I try fixed added, 
**0 */1 * * * ( /usr/lib/cgi-bin/ecu/startCoordinator.pl normal=1 )** to crontab.
why i saw this
14:06:03 checking zb system failed, need start (6700) command
14:06:18 checking zb system failed, no answer on serial
14:06:31 checking zb system running ok
14:06:58 polled inv 0 ts: 411 pw: 915.7
14:07:00 polled inv 1 ts: 411 pw: 1024.5
14:07:02 polled inv 2 ts: 411 pw: 1025.3
...