patience4711/RPI-APS-inverters #37: is it possible to control the inverter with this???

hschraep, 2022-05-06:
Hello,
find you project and its greate. Intresting would be if it is also possible to send commands to the inverter to let control him.
One option would be to let reduce the power of the inverter to let charge with the not needed power a battery. Would be great if this would be possible. rgds zizyfuzz

patience4711, 2022-05-06:
@hschraep The RPI-ECU has a console that allows the user to send zigbee commands. So in theory this is possible, you only have to find out what capabilities the inverter has and what the zigbee commands are. Currently nobody knows.

hschraep, 2022-05-06:
btw, all DS3 are compatible to your FW? DS3-S and also the DS3L?

patience4711, 2022-05-09:
@hschraep  I think so but i can not test this, i don't own a DS type inverter.

closed 2022-05-09 by patience4711