patience4711/RPI-APS-inverters #39: often fails to connect with inverters

b31k, 2022-05-19:
Hello good afternoon, the case is that it does not connect to the inverters on many days. If I force a poll from the console then it works fine, but in automatic mode it often fails, can you think why it can happen?

EVENTS JOURNAL

2022-05-19 01:00:00 running midNight.cgi
00:00:03 midNight.cgi moved energy total into db invEnergy
midNight: removed data files
sun.py : calculated sunrise / sunset
00:04:03 checking zb system failed, need start (6700) command
00:04:16 checking zb system running ok
06:40:06 woke up from sleep, goodmorning!
going to perform the morning ritual
invdataNull.cgi: wrote a new /ramdisk/invData0 with null values
invdataNull.cgi: wrote a new /ramdisk/invData1 with null values
invdataNull.cgi: wrote a new /ramdisk/invData2 with null values
Wrote 0 in database invData,
1st poll in 5 min, now exit
06:45:07 polled inv 0 error 10 AF_DATA_REQ fail
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData0
06:45:09 polled inv 1 error 10 AF_DATA_REQ fail
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData1
06:45:10 polled inv 2 error 10 AF_DATA_REQ fail
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData2
06:50:07 polled inv 0 error 10 AF_DATA_REQ fail
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData0
06:50:08 polled inv 1 error 10 AF_DATA_REQ fail
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData1
06:50:10 polled inv 2 error 10 AF_DATA_REQ fail
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData2
06:55:07 polled inv 0 error 10 AF_DATA_REQ fail
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData0
06:55:08 polled inv 1 error 10 AF_DATA_REQ fail
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData1
06:55:10 polled inv 2 error 10 AF_DATA_REQ fail
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData2
07:00:07 polled inv 0 error 10 AF_DATA_REQ fail
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData0
07:00:08 polled inv 1 error 10 AF_DATA_REQ fail
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData1
07:00:10 polled inv 2 error 10 AF_DATA_REQ fail
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData2
07:05:07 polled inv 0 error 10 AF_DATA_REQ fail
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData0
07:05:08 polled inv 1 error 10 AF_DATA_REQ fail
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData1
07:05:10 polled inv 2 error 10 AF_DATA_REQ fail
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData2
07:10:07 polled inv 0 error 10 AF_DATA_REQ fail
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData0
07:10:08 polled inv 1 error 10 AF_DATA_REQ fail
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData1
07:10:10 polled inv 2 error 10 AF_DATA_REQ fail
invdataNull.cgi: set values to null except
[... 114 lines cut ...]
for totalen and time in /ramdisk/invData1
08:15:10 polled inv 2 error 10 AF_DATA_REQ fail
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData2
08:20:07 polled inv 0 error 10 AF_DATA_REQ fail
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData0
08:20:09 polled inv 1 error 10 AF_DATA_REQ fail
invdataNull.cgi: set values to null except
....and so on indefinitely...

**but if i send command manually by console**

**16:27:36 checking zb system failed, need start (6700) command
16:27:50 checking zb system running ok
16:28:11 polled inv 0 the penultimate total energy value was 0
16:28:12 polled inv 0 ts: 34742 pw: 524.3
16:28:23 polled inv 1 the penultimate total energy value was 0
16:28:23 polled inv 1 ts: 34844 pw: 610.1
16:28:32 polled inv 2 the penultimate total energy value was 0
16:28:33 polled inv 2 ts: 34861 pw: 608.7**
[image]
[image]

thank you

patience4711, 2022-05-26:
@b31k  Thank you for your interest in this software....
Does this happen every morning or only when the esp/zigbee has been power-cycled?
I noticed that the zigbee module was started with the 2600 command.  In some cases a powercycle means that the zigbee network is cleared so when started with 2600 it doesn't recover.
I'l make a software update to repair that.

b31k, 2022-05-26:
The truth is that it is quite random.
Not all mornings
Thank you

patience4711, 2022-05-26:
@b31k You are the only one reporting this. With the information you provided it is hard to troubleshoot. Bad powersupply, range, i don't know.