patience4711/RPI-APS-inverters #42: CC2531 not detected anymore on linux after Flashing  with CC2530ZNP-with-SBL.hex

nlienard, 2022-06-04:
Hi
Before flashing, i can see on my rpi4/debian 10 the CC2531:

```
root@rpi4-cloud:~/CC2531/ESP-ECU-9_4/CC25xxfirmware ds3# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 007: ID 0451:16a2 Texas Instruments, Inc. CC Debugger
Bus 001 Device 008: ID 0451:16a8 Texas Instruments, Inc. CC2531 ZigBee
Bus 001 Device 003: ID 152d:2329 JMicron Technology Corp. / JMicron USA Technology Corp. JM20329 SATA Bridge
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub

root@rpi4-cloud:~/CC2531/ESP-ECU-9_4/CC25xxfirmware ds3# ls -la /dev/ttyACM0
crw-rw---- 1 root dialout 166, 0 Jun  4 13:17 /dev/ttyACM0

```

Flashing goes good:

```
root@rpi4-cloud:~/CC2531/ESP-ECU-9_4/CC25xxfirmware ds3# cc-tool -e -w  CC2530ZNP-with-SBL.hex
  Programmer: CC Debugger
  Target: CC2531
  Erasing flash...
  Completed
  Writing flash (241 KB)...
  Completed (17.95 s.)
```

But linux doesn't see the dongle anymore, is that normal ?

```
root@rpi4-cloud:~/CC2531/ESP-ECU-9_4/CC25xxfirmware ds3# ls -la /dev/ttyACM0
ls: cannot access '/dev/ttyACM0': No such file or directory
```

lsusb doesn't it anymore

```
root@rpi4-cloud:~/CC2531/ESP-ECU-9_4/CC25xxfirmware ds3#  lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 007: ID 0451:16a2 Texas Instruments, Inc. CC Debugger
Bus 001 Device 003: ID 152d:2329 JMicron Technology Corp. / JMicron USA Technology Corp. JM20329 SATA Bridge
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

If i reversed the firmware with CC2531ZNP-Prod.hex from zigbee2mqtt, it is detected back as well.

thanks

kadzsol, 2022-06-04:
Correct/Expected behaviour. The CC2530 does not have an USB interface.

nlienard, 2022-06-04:
Thanks !!
Now i want to use your WebGUI on my existing Raspberry 4 which is already running Debian.
I don't see any Howto... do you have a quick&dirty doc for that ? even without any comment, it would help.

For now, i'm trying to mount the ISO from the ZIP of this git. 

The IMG is a DOS image with a linux partition:

```
root@rpi4-cloud:/stuff/DS3L# fdisk -l RPI-ECU-v2_4.img
Disk RPI-ECU-v2_4.img: 4.37 GiB, 4694474752 bytes, 9168896 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x2189e003

Device            Boot  Start     End Sectors  Size Id Type
RPI-ECU-v2_4.img1        8192  532480  524289  256M  c W95 FAT32 (LBA)
RPI-ECU-v2_4.img2      540672 9168895 8628224  4.1G 83 Linux
```

Then i mounted it: 

```
root@rpi4-cloud:/stuff/DS3L# mount  -o loop,offset=276824064 RPI-ECU-v2_4.img /mnt
root@rpi4-cloud:/stuff/DS3L#
```

I copied all /var/www/html into my raspberry and the web gui is working but i need also all the backend package (mosquito, influxdb, grafana,  etc) .. that's why i'm asking if you have any quick&dirty doc to go quickly :)

I'll receive my DS3L on monday 17th june, so i'm preparing the land :)

thanks again !

patience4712, 2022-06-06:
No quick, and dirty, you have to burn the image on an sd card.

nlienard, 2022-06-06:
oh i see... about the CC2531, if there is no usb interface, how does it communicate ? thanks

patience4711, 2022-06-07:
@nlienard via the serial port on the connector pins. Please read here : https://github.com/patience4711/read-APSystems-YC600-QS1-DS3/issues/26

closed 2022-06-10 by patience4711

nlienard, 2022-06-10:
this link is about an ESP, i've a RPI. How can i know the GPIO connector to use on RPI ?
is there a file that i can modify to reflect the GPIO i ll use ?
thanks

peter33826, 2022-06-13:
See here
https://www.marrold.co.uk/2019/09/flashing-cc2531-zigbee-dongle-with.html
and
https://github.com/Koenkk/Z-Stack-firmware/issues/241

nlienard, 2022-06-13:
great, i'll try the schema connectivity from the blog article which looks clear.
I got the debug cable, so i'll use like him, and will let you know if it works.
thanks

nlienard, 2022-06-14:
i just installed your image on a SD CARD on a rpi 3+ and did the wiring as mentionned in the blog, but when running cc_chipid it doesn't work:

 ```
root@rpiecu:~/flash_cc2531# ./cc_chipid
  ID = 0000.
 ```

the wiring is here: https://imgur.com/a/rZvsUXI

what could be wrong ?

thanks

nlienard, 2022-06-14:
On https://www.marrold.co.uk/2019/09/flashing-cc2531-zigbee-dongle-with.html, it says to connect 4 cables, but on the other links, it is 5 cables, i don't understand :(

I did:

 Debug Connector | Raspberry Pi GPIO
GND | 1 | 39
RESET | 4 | 35
DC | 2 | 36
DD | 7 | 38

nlienard, 2022-06-14:
i guess my wiring is ok now because i managed to flash the CC2531 as well:

```
root@rpiecu:~# ./flash_cc2531/cc_chipid
  ID = b524.
```

```
root@rpiecu:~/CC25xxfirmware ds3# ../flash_cc2531/cc_erase
  ID = b524.
  erase result = 00a2.
root@rpiecu:~/CC25xxfirmware ds3# ../flash_cc2531/cc_write  CC2530ZNP-with-SBL.hex
  ID = b524.
  reading line 15490.
  file loaded (15497 lines read).
writing page 128/128..
 flash OK.
```

```
root@rpiecu:~/CC25xxfirmware ds3# ../flash_cc2531/cc_chipid
  ID = b524.
```

but when running "CheckCoordinator.pl", it doesn't see the CC2531 device, any clue ?

```
root@rpiecu:~/CC25xxfirmware ds3# /usr/lib/cgi-bin/ecu/checkCoordinator.pl
Content-Type: text/html; charset=ISO-8859-1

<html><head></head><body><span style='font-size:12px; font-family: arial;'>running script checkCoordinator.pl...<br>
calling checkCoordinator.cgi
command to run = sudo /usr/lib/cgi-bin/ecu/checkCoordinator.cgi<br>Content-type:text/html

<html>
<head>
<title>RPI ECU PROGAM</title>
</head>
<body>
<h4>running checkCoordinator.cgi</h4>
wiringPi started
<br>
opened port serial0
<br>
command = 00270027
<br>
no answer
<br>
no answer on serial
<br>
toLog = <br>13:53:36 checking zb system failed, no answer on serial
<br>
logCmd = echo "<br>13:53:36 checking zb system failed, no answer on serial" >> /ramdisk/ecu_log.txt
<br>
command runned<br>HTTP:1/1 200 OK
```

nlienard, 2022-06-14:
For now, i got this, it flashs good but now how to communicate ? i see it requires P0.2 and P0.3 but where to plug them on RPI GPIO ? and should i remove other connector ?

[image]

nlienard, 2022-06-14:
For operation, i guess the wiring is different, i don't see the GPIO number on the wiki, could you confirm this schema is good ?

[image]

thanks

nlienard, 2022-06-16:
IT WORKS !!! Schema is GOOD;  one cable (P0.2) was not plugged correctly on GPIO of RPI3

[image]