patience4711/RPI-APS-inverters #43: CC25

rasor89, 2022-06-04:
Before opening an issue please consider this: This project works with the described hardware as fas as i could test it. So some feedback is always welcome. Please state what hardware you use, raspberry platform, kind of inverter(s) you have and how many.

patience4712, 2022-06-06:
@rasor89  Thank you for your interest in this software.
I get the idea that you are flashing z-stack on your zb module. this wont work

rasor89, 2022-06-06:
@patience4712 I only used this guide for the Single steps, but used the hex files from your download.
Can you Tell me correct gpios and Pins on the CC2530 Module and how you did the flashing? Is there anything I have configurate in the raspberry in Addition? (config files or something like that?)

patience4712, 2022-06-06:
I flashed with the method of blog.boris. You should have wiring-PI updated. Otherwise i wouldn't know. Today i have no time to help you with this.

rasor89, 2022-06-06:
@patience4712 but the connection described in Blog Boris is different from what is described in your readme. This is why I Asked what is the correct one?

patience4711, 2022-06-07:
@rasor89 In my readme there is no reference to wiring for the purpose of flashing, only for the serial connection. 
For flashing the cc25xx i only referenced to the web where a lot is written on this subject.

closed 2022-06-07 by rasor89