patience4711/RPI-APS-inverters #44: Wrong Values

rasor89, 2022-06-12:
@patience4711 after your great Support the System is running. Unfortunately not in the way expected.
The values are shown can't comply to the real ones. I have the DS3L working togethet with a 460w panel.
I had this values from the beginning. Because I wasn'r Sure, I wiped the system and flashed new. But still with the same result. 
Especially I can't understand the curve for power with the high peaks and curve near to Zero. 
It was a sunny day without clouds. As per my electricity meter the energy has to be most of the time around 300-400w. Do you have any ideas what goes wrong? 
[image]
[image]
[image]

rasor89, 2022-06-13:
@patience4711
During my first try, I Updated to the latest version, but had same issues. That's why I tried this time first with original 2.4 Version, to be sure, the Update isn't the reason. But I will do the Updates this evening. 
No I didn't Do a factory reset after New Installation, but I started with completly empty (formated) sd card. Selected Inverter is ds3 (I have ds3-l) and also only one panel is installed. 
[image]

Following the result of the Poll. (I removed parts of the serial) 
`running script inverterPoll.pl...
argument is 0
sudo /usr/lib/cgi-bin/ecu/inverterPoll.cgi 0
Content-type:text/html
running inverterPoll.cgi
inverterpoll running with arg 0
invSerial from file = 7030xxxxxxxxx
invType = 2
idx = 123
invIdrev = 67BB
idShort = xxxx
idinvers from file = 80971B01B3D6
poll cmd = 1D240167BB1414060001000F1380971B01B3D6FBFB06BB000000000000C1FEFEEB
answer = FE0164010064FE034480001401D2FE0345C467BB005EFE0345C467BB005EFE7D44810000060167BB1414004800C011D3000069703000030249FBFB5CBBBB20000200E6FFFF0000000000008000068D02DE00450001036F138509780013001FFFFF05290866001F072C00000A3AFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF36D3FEFE67BB0246
messageToDecode = FE0164010064FE034480001401D2FE0345C467BB005EFE0345C467BB005EFE7D44810000060167BB1414004800C011D3000069703000030249FBFB5CBBBB20000200E6FFFF0000000000008000068D02DE00450001036F138509780013001FFFFF05290866001F072C00000A3AFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF36D3FEFE67BB0246
message split after 44810000 060167BB1414004800C011D3000069703000030249FBFB5CBBBB20000200E6FFFF0000000000008000068D02DE00450001036F138509780013001FFFFF05290866001F072C00000A3AFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF36D3FEFE67BB0246
json from file = { 'heath': 18.7, 'dcv': [ 15.1, 34.8, 0.0, 0.0 ], 'freq': 50.0, 'totalen': 37.5, 'acv': 231.1, 'en': [ 37.5, 0.0, 0.0, 0.0 ], 'dcc': [ 0.0, 0.0, 0.0, 0.0 ], 'idx': 123, 'totalpw': 0.4, 'pwr': [ 0.4, 0.0, 0.0, 0.0 ], 'time': 4150 }
time from file = 4150
en_old_total from file = 37.5
time from inverter 4150
the timespan = 4150
* * * * POLLED INVERTER 0 * * * *
** values panel 0**
extracted energy_ext[x] = 2033452.000000
en_new[x] = 37.618862 en_old[x] = 37.500000
en_increase[x] = 37.618862
panel 0 energy to now 37.6189
no panel 1
no panel 2
no panel 3
toLog =
18:35:24 polled inv 0 inverter restarted
logCmd = echo "
18:35:24 polled inv 0 inverter restarted" >> /ramdisk/ecu_log.txt
en_new_total: 37.6189
en_increase_total: 37.6189
echo "{ 'totalpw': 0.4, 'en': [ 37.6, 0.0, 0.0, 0.0 ], 'dcc': [ 0.0, 0.0, 0.0, 0.0 ], 'pwr': [ 0.4, 0.0, 0.0, 0.0 ], 'time': 4150, 'idx': 123, 'acv': 231.3, 'totalen': 37.6, 'freq': 50.0, 'dcv': [ 15.3, 34.9, 0.0, 0.0 ], 'heath': 18.7 }" > "/ramdisk/invData0"
sedCmd = sed -i 's|['\'']|'\"'|g' /ramdisk/invData0
dbCmd = curl -i -XPOST 'http://localhost:8086/write?db=invData' --data-binary 'inv0 p=0.4,e=37.6'
HTTP/1.1 204 No Content Content-Type: application/json Request-Id: d387853f-eb36-11ec-81d8-b827eb77c43f X-Influxdb-Build: OSS X-Influxdb-Version: 1.8.10 X-Request-Id: d387853f-eb36-11ec-81d8-b827eb77c43f Date: Mon, 13 Jun 2022 16:35:25 GMT all cmds sent
returned by sendCommands() 0
inverterpoll exit with code 0
toLog =
18:35:25 polled inv 0 ts: 4150 pw: 0.4
logCmd = echo "
18:35:25 polled inv 0 ts: 4150 pw: 0.4" >> /ramdisk/ecu_log.txt
command processed
HTTP:1/1 200 OK `

patience4711, 2022-06-13:
Oke, this looks like a first poll but it looks allright at first sight. Normally the timespan should be 300 seconds, but is 4150. 
That is wrong so the calculations are over a wrong timespan
**time from file = 4150
time from inverter 4150
the timespan = 4150**
I made a new softwareupdate to 2_8 in which this bug is repaired. 
Later this evening i wil provide a new image of 2_8, maybe preferable.

rasor89, 2022-06-13:
@patience4711 great. I will try 2_8 and let you know the result.
At the moment I try to reflash the system with some problems. Wifi get some seconds connected to the configurated one, then the connection gets lost and RadioAP is revisible. wpa_supplicant.conf seems correct.

patience4711, 2022-06-14:
It seems that you are the first one who works with this sw and the ds3 otherwise we would have found this bug earlier. 
I have a new sd image ready but first i have to be sure that it works now with the 2_8 upgrade.  I dont have a ds3 so i can't test anything myself.

patience4711, 2022-06-14:
I found another bug as to the calculation of the power.
If you download and install the file ecu_swupdate_14-06-2022.tar below, it will work as it should.
[ ecu_swupdate_14-06-2022.tar](https://1drv.ms/u/s!AkMG5FvUwhedi0w0itFjZ4A_Jc-n?e=bhabdM)

rasor89, 2022-06-15:
I installed the update yesterday late afternoon, I didn't get a chart anymore, but from the first view, it seems ok. Let me check this afternoon again, then I can close the issue. **Your support and fast reaction and fixing is great.**

Besides: MQTT isn't working for me, I will open a new topic to disuss about.

rasor89, 2022-06-15:
So It's seems It's working well! 
[image]

Is it possible to add a 3rd graph in the diagram which shows total generated energie of the Day? Same as in the table? 
[image]
[image]

closed 2022-06-16 by rasor89