patience4711/RPI-APS-inverters #46: Is MQTT working proper?

rasor89, 2022-06-15:
@patience4711 I'm not sure, if MQTT is working proper, or whether the failure is on my side, but I didn't get it working.

First question: for the Koenkk/zigbee2mqtt there is the option to get a json output via an url 
`http://IP.OF.ESP.ECU/get.Inverter?inv=x`
As I'm right, this isn't the case here, and I have to use MQTT via a client?

I enabled the MQTT in the menü. For the IP of the broker "localhost" should be working, but i tested also with the IP of the broker (also not successful).  
My MQTT broker has an user with password. When I fill in the username and the passwort and save the changes, the information disappear. Both cells are empty after saving it. Is that correct? Because when I'm testing the MQTT, there is no username or password in the command. See screenshot attached. Can you please check this? Thanks!

patience4711, 2022-06-15:
@rasor89  The mqtt works but the password is not used. (and saved).  I always use the ip because my broker runs on another raspberry. Maybe remove the authentication, it is not very usefull inside your network.

The data request link, where did you read that? That is a feature of my esp implementation. 
Comming 7 days i have no time for maintenance...

rasor89, 2022-06-15:
@patience4711 what do you mean with the password is not used? So it didn't work? Ok I will try without authentification.

Read that in another forum (for KNX), where a user told about the ESP implementation. So maybe the reference to Koenkk/zigbee2mqtt isn't correct.

rasor89, 2022-06-16:
So I tried MQTT without authentification and it works.

closed 2022-06-16 by rasor89