patience4711/RPI-APS-inverters #48: Bad value on DS3L -

nlienard, 2022-06-17:
Hi

firmware version = RPI-ECU-v2_8 

I got a DS3L connected to a RPI3 with a CC2531 dongle, flashed as well with the firmware you provided.

Interface is working fine but reporting wrong value: 0.4W

[image]

I plugged a Wattmettre on the DS3L and i see at the same moment 25W

Any idea what could be wrong ?

[image]

[image]

thanks

nlienard, 2022-06-18:
this morning at 9h08

[image]

and with my wattmeter:

[image]

patience4711, 2022-06-20:
please read [here](https://github.com/patience4711/RPI-APS-inverters/wiki/3-INVERTERS#polling) for some information on the polling process and calculations.
If you are convinced that its wrong you could do the following:
Do a manual poll 2 minutes after a regular automatic poll (every 5 minutes) and make a screenshot 
like [see here](https://github.com/patience4711/RPI-APS-inverters/issues/44#issuecomment-1154149364) and than repeat this 5 minutes later. 
From this data i can investigate this.

nlienard, 2022-06-20:
thanks.

this is a first shot of

```
root@rpiecu:~# /usr/lib/cgi-bin/ecu/inverterPoll.cgi 0
Content-type:text/html

<html>
<head>
<title>RPI ECU PROGRAM</title>
</head>
<body>
<h4>running inverterPoll.cgi</h4>
inverterpoll running with arg 0
<br>
invSerial from file = 703000069723
<br>
invType = 2
<br>
idx = 0
<br>
invIdrev = E1D5
<br>
idShort  = xxxx
<br>
idinvers from file = 80971B01B3D6
<br>
poll cmd = 1D2401E1D51414060001000F1380971B01B3D6FBFB06BB000000000000C1FEFE03
<br>
 answer = FE0164010064FE034480E914013B
<br>
messageToDecode = FE0164010064FE034480E914013B
<br>
no AF_DATA_CONFIRM
<br>
all cmds sent
<br>
returned by sendCommands() 11
<br>
toLog = <br>15:27:31 polled inv 0 error 11 AF_DATA_CONFIRM fail
<br>
logCmd = echo "<br>15:27:31 polled inv 0 error 11 AF_DATA_CONFIRM fail" >> /ramdisk/ecu_log.txt
<br>
Content-type:text/html

<html>
<head>
<title>RPI ECU PROGRAM</title>
</head>
<body>
<h4>running invdataNull.cgi</h4>
echo "{
    'idx': 0,
    'totalen': 2185.9,
    'freq': 0,
    'heath': 0,
    'dcv': [
        0,
        0,
        0,
        0
    ],
    'time': 25206,
    'pwr': [
        0,
        0,
        0,
        0
    ],
    'acv': 0,
    'en': [
        1035.3,
        1150.6,
        0.0,
        0.0
    ],
    'dcc': [
        0,
        0,
        0,
        0
    ],
    'totalpw': 0
}" > "/ramdisk/invData0"
<br>
sedCmd = sed -i 's|['\'']|'\"'|g' /ramdisk/invData0
<br>
invdataNull.cgi: resetted existing invData
<br>
invdataNull.cgi exit with code 0
<br>
inverterpoll exit with code 0
<br>
```

[image]

nlienard, 2022-06-20:
```
root@rpiecu:~# /usr/lib/cgi-bin/ecu/autoPoll.pl
Content-Type: text/html; charset=ISO-8859-1

<html><head></head><body><span style='font-size:12px; font-family: arial;'>running script autoPoll.pl...<br>
current totalminutes = 930
pollingtime
filename=/var/www/ecu_data/inverters/invProperties0
<br>autopollCmd = sudo /usr/lib/cgi-bin/ecu/inverterPoll.cgi 0
<br>Content-type:text/html

<html>
<head>
<title>RPI ECU PROGRAM</title>
</head>
<body>
<h4>running inverterPoll.cgi</h4>
inverterpoll running with arg 0
<br>
invSerial from file = 703000069723
<br>
invType = 2
<br>
idx = 0
<br>
invIdrev = E1D5
<br>
idShort  = xxxx
<br>
idinvers from file = 80971B01B3D6
<br>
poll cmd = 1D2401E1D51414060001000F1380971B01B3D6FBFB06BB000000000000C1FEFE03
<br>
 answer = FE0164010064FE034480E914013B
<br>
messageToDecode = FE0164010064FE034480E914013B
<br>
no AF_DATA_CONFIRM
<br>
all cmds sent
<br>
returned by sendCommands() 11
<br>
toLog = <br>15:30:18 polled inv 0 error 11 AF_DATA_CONFIRM fail
<br>
logCmd = echo "<br>15:30:18 polled inv 0 error 11 AF_DATA_CONFIRM fail" >> /ramdisk/ecu_log.txt
<br>
Content-type:text/html

<html>
<head>
<title>RPI ECU PROGRAM</title>
</head>
<body>
<h4>running invdataNull.cgi</h4>
echo "{
    'heath': 0,
    'dcv': [
        0,
        0,
        0,
        0
    ],
    'freq': 0,
    'totalen': 2185.9,
    'idx': 0,
    'dcc': [
        0,
        0,
        0,
        0
    ],
    'en': [
        1035.3,
        1150.6,
        0.0,
        0.0
    ],
    'totalpw': 0,
    'acv': 0,
    'pwr': [
        0,
        0,
        0,
        0
    ],
    'time': 25206
}" > "/ramdisk/invData0"
<br>
sedCmd = sed -i 's|['\'']|'\"'|g' /ramdisk/invData0
<br>
invdataNull.cgi: resetted existing invData
<br>
invdataNull.cgi exit with code 0
<br>
inverterpoll exit with code 0
<br>
result = 0
<br>Content-type:text/html

<html>
<head>
<title>RPI ECU PROGAM</title>
</head>
<body>
<h4>running sendMqtt.cgi</h4>
sendMqtt.cgi running with arg 0<br>
wiringPi started
<br>
fileToread = /var/www/ecu_data/mqttConfig.json<br>
fileToread = /var/www/ecu_data/inverters/invProperties0
<br>
fileToread = /ramdisk/invData0<br>
toMqtt = '{"idx":0,"nvalue":0,"svalue":"0;2185.9"}'
<br>
systemCmd = mosquitto_pub -h 127.0.0.1 -p  1883 -t domoticz/in -m '{"idx":0,"nvalue":0,"svalue":"0;2185.9"}'
<br>
sent message format 1 domoticz
<br>
filename=/var/www/ecu_data/inverters/invProperties1
<br>filename not exists, skipping<
<br>filename=/var/www/ecu_data/inverters/invProperties2
<br>filename not exists, skipping<
<br>filename=/var/www/ecu_data/inverters/invProperties3
<br>filename not exists, skipping<
<br>filename=/var/www/ecu_data/inverters/invProperties4
<br>filename not exists, skipping<
<br>filename=/var/www/ecu_data/inverters/invProperties5
<br>filename not exists, skipping<
<br>filename=/var/www/ecu_data/inverters/invProperties6
<br>filename not exists, skipping<
<br>filename=/var/www/ecu_data/inverters/invProperties7
<br>filename not exists, skipping<
<br>filename=/var/www/ecu_data/inverters/invProperties8
<br>filename not exists, skipping<
<br>command processed
<br>HTTP:1/1 200 OKroot@rpiecu:~#
root@rpiecu:~# date ; /usr/lib/cgi-bin/ecu/inverterPoll.cgi 0
Mon 20 Jun 15:31:21 CEST 2022
Content-type:text/html

<html>
<head>
<title>RPI ECU PROGRAM</title>
</head>
<body>
<h4>running inverterPoll.cgi</h4>
inverterpoll running with arg 0
<br>
invSerial from file = 703000069723
<br>
invType = 2
<br>
idx = 0
<br>
invIdrev = E1D5
<br>
idShort  = xxxx
<br>
idinvers from file = 80971B01B3D6
<br>
poll cmd = 1D2401E1D51414060001000F1380971B01B3D6FBFB06BB000000000000C1FEFE03
<br>
 answer = FE0164010064FE034480E914013B
<br>
messageToDecode = FE0164010064FE034480E914013B
<br>
no AF_DATA_CONFIRM
<br>
all cmds sent
<br>
returned by sendCommands() 11
<br>
toLog = <br>15:31:22 polled inv 0 error 11 AF_DATA_CONFIRM fail
<br>
logCmd = echo "<br>15:31:22 polled inv 0 error 11 AF_DATA_CONFIRM fail" >> /ramdisk/ecu_log.txt
<br>
Content-type:text/html

<html>
<head>
<title>RPI ECU PROGRAM</title>
</head>
<body>
<h4>running invdataNull.cgi</h4>
echo "{
    'time': 25206,
    'pwr': [
        0,
        0,
        0,
        0
    ],
    'acv': 0,
    'totalpw': 0,
    'en': [
        1035.3,
        1150.6,
        0.0,
        0.0
    ],
    'dcv': [
        0,
        0,
        0,
        0
    ],
    'heath': 0,
    'freq': 0,
    'totalen': 2185.9,
    'idx': 0,
    'dcc': [
        0,
        0,
        0,
        0
    ]
}" > "/ramdisk/invData0"
<br>
sedCmd = sed -i 's|['\'']|'\"'|g' /ramdisk/invData0
<br>
invdataNull.cgi: resetted existing invData
<br>
invdataNull.cgi exit with code 0
<br>
inverterpoll exit with code 0
<br>
root@rpiecu:~#
```

nlienard, 2022-06-20:
```
root@rpiecu:~# date ; /usr/lib/cgi-bin/ecu/inverterPoll.cgi 0
Mon 20 Jun 15:36:12 CEST 2022
Content-type:text/html

<html>
<head>
<title>RPI ECU PROGRAM</title>
</head>
<body>
<h4>running inverterPoll.cgi</h4>
inverterpoll running with arg 0
<br>
invSerial from file = 703000069723
<br>
invType = 2
<br>
idx = 0
<br>
invIdrev = E1D5
<br>
idShort  = xxxx
<br>
idinvers from file = 80971B01B3D6
<br>
poll cmd = 1D2401E1D51414060001000F1380971B01B3D6FBFB06BB000000000000C1FEFE03
<br>
 answer = FE0164010064FE034480E914013B
<br>
messageToDecode = FE0164010064FE034480E914013B
<br>
no AF_DATA_CONFIRM
<br>
all cmds sent
<br>
returned by sendCommands() 11
<br>
toLog = <br>15:36:13 polled inv 0 error 11 AF_DATA_CONFIRM fail
<br>
logCmd = echo "<br>15:36:13 polled inv 0 error 11 AF_DATA_CONFIRM fail" >> /ramdisk/ecu_log.txt
<br>
Content-type:text/html

<html>
<head>
<title>RPI ECU PROGRAM</title>
</head>
<body>
<h4>running invdataNull.cgi</h4>
echo "{
    'en': [
        1035.3,
        1150.6,
        0.0,
        0.0
    ],
    'dcv': [
        0,
        0,
        0,
        0
    ],
    'heath': 0,
    'totalpw': 0,
    'acv': 0,
    'idx': 0,
    'dcc': [
        0,
        0,
        0,
        0
    ],
    'totalen': 2185.9,
    'freq': 0,
    'time': 25206,
    'pwr': [
        0,
        0,
        0,
        0
    ]
}" > "/ramdisk/invData0"
<br>
sedCmd = sed -i 's|['\'']|'\"'|g' /ramdisk/invData0
<br>
invdataNull.cgi: resetted existing invData
<br>
invdataNull.cgi exit with code 0
<br>
inverterpoll exit with code 0
<br>
```

patience4711, 2022-06-20:
All of this are failed polls, not usefull. Can you please remove them to keep this readable? You should get a poll answer, not FE0164010064FE034480E914013B. 
You can check the journal if there is a clou.
Maybe you are too far away from the inverter?

nlienard, 2022-06-20:
got this different answer, does it help ?

```
invIdrev = E1D5
<br>
idShort  = xxxx
<br>
idinvers from file = 80971B01B3D6
<br>
poll cmd = 1D2401E1D51414060001000F1380971B01B3D6FBFB06BB000000000000C1FEFE03
<br>
 answer = FE0164010064FE034480CD14011F
<br>
messageToDecode = FE0164010064FE034480CD14011F
<br>
no AF_DATA_CONFIRM
```

nlienard, 2022-06-20:
After moving a bit the antenna, got a better one:

```
<br>
invIdrev = E1D5
<br>
idShort  = xxxx
<br>
idinvers from file = 80971B01B3D6
<br>
poll cmd = 1D2401E1D51414060001000F1380971B01B3D6FBFB06BB000000000000C1FEFE03
<br>
 answer = FE0164010064FE034480001401D2FE7D448100000601E1D51414002400DFE60F004F69703000069723FBFB5CBBBB200003000FFFFF000000000000000006A806B5018B01E00358138A829B0158001DFFFF05200C3C05759AE7060D5BACFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF39B8FEFEE1D5D5AF
<br>
messageToDecode = FE0164010064FE034480001401D2FE7D448100000601E1D51414002400DFE60F004F69703000069723FBFB5CBBBB200003000FFFFF000000000000000006A806B5018B01E00358138A829B0158001DFFFF05200C3C05759AE7060D5BACFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF39B8FEFEE1D5D5AF
<br>
no route receipt
<br>message split after 44810000 0601E1D51414002400DFE60F004F69703000069723FBFB5CBBBB200003000FFFFF000000000000000006A806B5018B01E00358138A829B0158001DFFFF05200C3C05759AE7060D5BACFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF39B8FEFEE1D5D5AF
<br>
json from file = {
    'heath': 0,
    'dcv': [
        0,
        0,
        0,
        0
    ],
    'totalpw': 0,
    'freq': 0,
    'totalen': 2185.9,
    'dcc': [
        0,
        0,
        0,
        0
    ],
    'en': [
        1035.3,
        1150.6,
        0.0,
        0.0
    ],
    'idx': 0,
    'acv': 0,
    'time': 25206,
    'pwr': [
        0,
        0,
        0,
        0
    ]
}
```

nlienard, 2022-06-20:
another polling:

```
root@rpiecu:~# date ; /usr/lib/cgi-bin/ecu/inverterPoll.cgi 0
Mon 20 Jun 16:11:11 CEST 2022
Content-type:text/html

<html>
<head>
<title>RPI ECU PROGRAM</title>
</head>
<body>
<h4>running inverterPoll.cgi</h4>
inverterpoll running with arg 0
<br>
invSerial from file = 703000069723
<br>
invType = 2
<br>
idx = 0
<br>
invIdrev = E1D5
<br>
idShort  = xxxx
<br>
idinvers from file = 80971B01B3D6
<br>
poll cmd = 1D2401E1D51414060001000F1380971B01B3D6FBFB06BB000000000000C1FEFE03
<br>
 answer = FE0164010064FE034480001401D2FE0345C4E1D500B6FE0345C4E1D500B6FE7D448100000601E1D51414002200A53F1B000069703000069723FBFB5CBBBB200003000FFFFF000000000000000006A706B2019B01F0035713898389015F001DFFFF04F50C39057F4B430619365AFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF3945FEFEE1D57038
<br>
messageToDecode = FE0164010064FE034480001401D2FE0345C4E1D500B6FE0345C4E1D500B6FE7D448100000601E1D51414002200A53F1B000069703000069723FBFB5CBBBB200003000FFFFF000000000000000006A706B2019B01F0035713898389015F001DFFFF04F50C39057F4B430619365AFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF3945FEFEE1D57038
<br>
message split after 44810000 0601E1D51414002200A53F1B000069703000069723FBFB5CBBBB200003000FFFFF000000000000000006A706B2019B01F0035713898389015F001DFFFF04F50C39057F4B430619365AFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF3945FEFEE1D57038
<br>
json from file = {
    'heath': 38.1,
    'dcv': [
        35.8,
        35.6,
        0.0,
        0.0
    ],
    'freq': 50.0,
    'totalen': 3222.6,
    'acv': 225.5,
    'en': [
        1527.9,
        1694.7,
        0.0,
        0.0
    ],
    'dcc': [
        5.1,
        6.1,
        0.0,
        0.0
    ],
    'idx': 0,
    'totalpw': 4.8,
    'pwr': [
        2.1,
        2.6,
        0.0,
        0.0
    ],
    'time': 33604
}
<br>
time from file = 33604
<br>
en_old_total from file = 3222.6
<br>
time from inverter 33673
<br>
the timespan = 69
<br>
* * * * POLLED INVERTER 0 * * * *
<br>** values panel 0**
<br>
extracted energy_ext[x] = 92228416.000000
<br>
en_new[x] = 1530.991577  en_old[x] = 1527.900024
<br>
en_increase[x] = 3.091553
<br>
panel 0  energy to now 1530.99
<br>
** values panel 1**
<br>
extracted energy_ext[x] = 102315608.000000
<br>
en_new[x] = 1698.439087  en_old[x] = 1694.699951
<br>
en_increase[x] = 3.739136
<br>
panel 1  energy to now 1698.44
<br>
no panel 2
<br>
no panel 3
<br>
  en_new_total: 3229.43
<br>
  en_increase_total: 6.83069
<br>
echo "{
    'totalpw': 4.8,
    'en': [
        1531.0,
        1698.4,
        0.0,
        0.0
    ],
    'dcc': [
        5.1,
        6.2,
        0.0,
        0.0
    ],
    'pwr': [
        2.2,
        2.6,
        0.0,
        0.0
    ],
    'time': 33673,
    'idx': 0,
    'acv': 225.0,
    'totalen': 3229.4,
    'freq': 50.0,
    'dcv': [
        35.7,
        35.5,
        0.0,
        0.0
    ],
    'heath': 38.1
}" > "/ramdisk/invData0"
<br>
sedCmd = sed -i 's|['\'']|'\"'|g' /ramdisk/invData0
<br>
dbCmd = curl -i -XPOST 'http://localhost:8086/write?db=invData' --data-binary 'inv0 p=4.8,e=6.8'
<br>
HTTP/1.1 204 No Content
Content-Type: application/json
Request-Id: d713ac82-f0a2-11ec-8032-b827eb14e27a
X-Influxdb-Build: OSS
X-Influxdb-Version: 1.8.10
X-Request-Id: d713ac82-f0a2-11ec-8032-b827eb14e27a
Date: Mon, 20 Jun 2022 14:11:12 GMT

all cmds sent
<br>
returned by sendCommands() 0
<br>
inverterpoll exit with code 0
<br>
toLog = <br>16:11:12 polled inv 0 ts: 69 pw: 4.8
<br>
logCmd = echo "<br>16:11:12 polled inv 0 ts: 69 pw: 4.8" >> /ramdisk/ecu_log.txt
<br>
root@rpiecu:~#
```
[image]

patience4711, 2022-06-20:
It looks that there are no values extracted from the poll answer. So this answer is not correct. The last one looks good.

nlienard, 2022-06-20:
Do you need other polling values ?
which variable are you looking at exactly ?

nlienard, 2022-06-20:
I took a picture of the wattmeter in same moment of the polling on RPI:

[image]

Polling values output :

```
root@rpiecu:~# date ; /usr/lib/cgi-bin/ecu/inverterPoll.cgi 0
Mon 20 Jun 16:42:08 CEST 2022
Content-type:text/html

<html>
<head>
<title>RPI ECU PROGRAM</title>
</head>
<body>
<h4>running inverterPoll.cgi</h4>
inverterpoll running with arg 0
<br>
invSerial from file = 703000069723
<br>
invType = 2
<br>
idx = 0
<br>
invIdrev = E1D5
<br>
idShort  = xxxx
<br>
idinvers from file = 80971B01B3D6
<br>
poll cmd = 1D2401E1D51414060001000F1380971B01B3D6FBFB06BB000000000000C1FEFE03
<br>
 answer = FE0164010064FE034480001401D2FE0345C4E1D500B6FE0345C4E1D500B6FE7D448100000601E1D51414002400C6C973000069703000069723FBFB5CBBBB200003000FFFFF000000000000000006D806AC0114018C036013858AC90104001EFFFF04FA0C0A05BFB129066B2144FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF3900FEFEE1D57029
<br>
messageToDecode = FE0164010064FE034480001401D2FE0345C4E1D500B6FE0345C4E1D500B6FE7D448100000601E1D51414002400C6C973000069703000069723FBFB5CBBBB200003000FFFFF000000000000000006D806AC0114018C036013858AC90104001EFFFF04FA0C0A05BFB129066B2144FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF3900FEFEE1D57029
<br>
message split after 44810000 0601E1D51414002400C6C973000069703000069723FBFB5CBBBB200003000FFFFF000000000000000006D806AC0114018C036013858AC90104001EFFFF04FA0C0A05BFB129066B2144FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF3900FEFEE1D57029
<br>
json from file = {
    'heath': 37.2,
    'dcv': [
        35.6,
        36.5,
        0.0,
        0.0
    ],
    'freq': 50.0,
    'totalen': 3385.6,
    'acv': 226.3,
    'en': [
        1599.8,
        1785.8,
        0.0,
        0.0
    ],
    'dcc': [
        3.5,
        4.9,
        0.0,
        0.0
    ],
    'idx': 0,
    'totalpw': 3.6,
    'pwr': [
        1.5,
        2.1,
        0.0,
        0.0
    ],
    'time': 35489
}
<br>
time from file = 35489
<br>
en_old_total from file = 3385.6
<br>
time from inverter 35529
<br>
the timespan = 40
<br>
* * * * POLLED INVERTER 0 * * * *
<br>** values panel 0**
<br>
extracted energy_ext[x] = 96448808.000000
<br>
en_new[x] = 1601.050049  en_old[x] = 1599.800049
<br>
en_increase[x] = 1.250000
<br>
panel 0  energy to now 1601.05
<br>
** values panel 1**
<br>
extracted energy_ext[x] = 107684160.000000
<br>
en_new[x] = 1787.556885  en_old[x] = 1785.800049
<br>
en_increase[x] = 1.756836
<br>
panel 1  energy to now 1787.56
<br>
no panel 2
<br>
no panel 3
<br>
  en_new_total: 3388.61
<br>
  en_increase_total: 3.00684
<br>
echo "{
    'totalpw': 3.6,
    'en': [
        1601.1,
        1787.6,
        0.0,
        0.0
    ],
    'dcc': [
        3.5,
        4.9,
        0.0,
        0.0
    ],
    'pwr': [
        1.5,
        2.1,
        0.0,
        0.0
    ],
    'time': 35529,
    'idx': 0,
    'acv': 227.4,
    'totalen': 3388.6,
    'freq': 50.0,
    'dcv': [
        35.6,
        36.5,
        0.0,
        0.0
    ],
    'heath': 37.2
}" > "/ramdisk/invData0"
<br>
sedCmd = sed -i 's|['\'']|'\"'|g' /ramdisk/invData0
<br>
dbCmd = curl -i -XPOST 'http://localhost:8086/write?db=invData' --data-binary 'inv0 p=3.6,e=3.0'
<br>
HTTP/1.1 204 No Content
Content-Type: application/json
Request-Id: 29c7dfdc-f0a7-11ec-803d-b827eb14e27a
X-Influxdb-Build: OSS
X-Influxdb-Version: 1.8.10
X-Request-Id: 29c7dfdc-f0a7-11ec-803d-b827eb14e27a
Date: Mon, 20 Jun 2022 14:42:09 GMT

all cmds sent
<br>
returned by sendCommands() 0
<br>
inverterpoll exit with code 0
<br>
toLog = <br>16:42:09 polled inv 0 ts: 40 pw: 3.6
<br>
logCmd = echo "<br>16:42:09 polled inv 0 ts: 40 pw: 3.6" >> /ramdisk/ecu_log.txt
<br>
```

nlienard, 2022-06-20:
at the moment i installed, there were not the 2_8 available. 
so I installed 2_4 then i did one by one all the update until the final one into 2_8 and didn't see any error on the web gui.

nlienard, 2022-06-20:
Ok done

[image]

then i ran the polling and now it looks GOOD 👍 
[image]

what was the issue ? i failed to update to latest ?

patience4711, 2022-06-20:
You had the wrong version 2_8, i issued 2 versions 8 shortly after each other, expecting that it wouldn't go wrong.  So you probably picked the first one accidently.

closed 2022-06-22 by patience4711