patience4711/RPI-APS-inverters #50: Inverter restart resets daily energy

rasor89, 2022-06-19:
@patience4711 I noticed that my Inverter restarts around 6pm, but don't know the reasons (Power is still high enough). Whatever, the ecu resets the daily energy in the stats. Total energy will be summarized in the charts, but not in the overview and in the MQTT message. Can you investigate the reasons, please? Is assume this is a topic of the ecu ans not the invester itself. Ist's more usefull to not reset the values other time than midnight. 

Besides: how can I reduce the poll time? I want to try to reduce to 1min.

[image]
[image]
[image]

patience4711, 2022-06-20:
How this works you can read [here](https://github.com/patience4711/RPI-APS-inverters/wiki/3-INVERTERS#polling)

Very strange that your inverter restarts at the same time every day. There must be a reason for that.

rasor89, 2022-06-20:
@patience4711 I noticed it's not always the same time. Yesterday it was 10 min earlier. But as you can see from the journal, this is not reasoned by the missing sunlight. 
Ist the calculation done by the inverter? I thought the ECU is calculating? So there is no possibilty to change this? Besides the restart, on cloudy days, the energy will always be reseted.

Which files I have to modify to reduce the poll time down to 1 min?

patience4711, 2022-06-20:
The polling is triggered by the cron so #crontab -e . I wouldn't do this, it may lead to crash of the zigbee radio.
In the wiki i explained the calculation. [wiki](https://github.com/patience4711/RPI-APS-inverters/wiki/3-INVERTERS#polling)

rasor89, 2022-06-20:
@patience4711 Do you think It's possible to add a timestamp to the MQTT message?

patience4711, 2022-06-21:
It is possible but if you subscribe you can do mosquitto_sub -v -t /# | ts 
so the subscriber always get the timestamp. see moreutils package.

I have been working on a new feature: triggering a poll via a http request. This way you can poll in a higher frequency.
You have to disable the autopolling in the settings. Read the help in the settings for how to.

you can download it [ecu_swupdate_21-06-2022](https://1drv.ms/u/s!AkMG5FvUwhedi07WuKwBzTh8g9pk?e=baYbMG)

rasor89, 2022-06-21:
Oh I saw you're working on a http request. So there is something coming?

patience4711, 2022-06-21:
Yes, install this upgrade [ecu_swupdate_21-06-2022](https://1drv.ms/u/s!AkMG5FvUwhedi08t2z6W3zYg0PvF?e=gGXX8G)<br><br>
Please look at the help page in 'menu->settings'

patience4711, 2022-06-22:
I also added a http request for data: 'ip_of_ecu/recovery/getInverter.php?inv=x' returnes a json with the data
[ecu_swupdate_22-06-2022](https://1drv.ms/u/s!AkMG5FvUwhedi1HjxseRKpPVfKge?e=CIYbN8)

nlienard, 2022-06-22:
having similar issue, the inverter restarted at 19h18 and second time at 20h42.
What could be the reason ?
I noticed also lot of message "error 11 AF_DATA_CONFIRM fail"
Could it be communication issue ?
I had to modify cron to run a startCoordinator because it was failing too many time  yesterday, i guess i'm a bit far from the inverter and sometimes it fails to communicate. By doing a StartCoordinator, it gave me better results but don't know if this is bad. Also i put every 2 mn, is it bad ? i i'm using the data through mqtt for Home Assistant.

```
*/2 * * * *   ( /usr/lib/cgi-bin/ecu/startCoordinator.pl normal=1 ) ;   (/usr/lib/cgi-bin/ecu/autoPoll.pl)
```

[image]

[image]

[image]

Chart looks ok

[image]

any clue is welcome.

rasor89, 2022-06-22:
Installed the Update from yesterday, but didn't use the new feature.
Today I didn't habe a restart, but a lot of error messages, Same as nlienard.
After that I no New values arrived. 
[image]

nlienard, 2022-06-23:
it is possible to add a condition in autopoll.pl (or  maybe inverterPoll.cgi) which checks if totalpw variable is empty and relaunch another polling.
It happens on my side that when running a first manual polling , it responds empty value at 0 and just after another manual polling, and i get the value as well.
thanks

patience4711, 2022-06-23:
Like i said before, polling more than once every 5 minutes leads to instability so i am not going to support that.

In autoPoll.pl this check is there, in case of 0 the poll is ignored (see the journal). Manual poll does not use autopoll. Autopoll keeps running to check if the sun is set or not. 

When there is an time value polled which is lower than the previous one, the program assumes that the inverter is reset. Otherwise lower values are not possible. 
Due to instability sometimes eratical values may pop-up which could lead to this assumption.

The radio trafic failes a lot in your cases, that must be a range problem, maybe a better radio should be used. The cc2531 is not the best one.
When the reset happens around the same time, there is a systematical reason for that, you say it is not the sunlight. Well i can't think of any other reason but i can't help here. Maybe ask the manufacturer.

The subject of this issue is inverter resets, to improve the issues as a knowledge base, please open another issue, like "polling resolution"  or so, and move your relative posts to there. Thanks

rasor89, 2022-06-23:
@patience4711 I noticed your concerns about polling less than 5 Minutes. So I didn't change. In moved the CC2530 yesterday 10cm, after that it proper worked for more than 1h. Then I got the above failures. There Was so other change in the configuration.
Is there recommendation for a netter Module than the CC2530 / CC2531 to improve the range?

patience4711, 2022-06-23:
I have about the same conditions, 6mtr and only a door with glass betweeen The only time i saw some disturbances in the zigbee trafic was when i moved my body between the zb module (2530) and the inverter. So it is sensitive to this.
I don't know how you wired it, the dupontcables are not very reliable so i soldered the wiring. Suspect also your powersupply, and keep the radio away from it.  Or try with another one.
I am afraid i can't solve this riddle.

kadzsol, 2022-06-23:
> @patience4711 I noticed your concerns about polling less than 5 Minutes. So I didn't change. In moved the CC2530 yesterday 10cm, after that it proper worked for more than 1h. Then I got the above failures. There Was so other change in the configuration.
> 
> Normally the range shouldn't be a problem. I have only 5m between my Raspberry and the inverter with a window in between,

You may want to try a CC2530 with an amplifier chip (CC2530+CC2590/1/2)

nlienard, 2022-06-23:
inverter just rebooted again

```
<br>18:26:02 checking zb system running ok
<br>18:26:04 polled inv 0 ts: 120 pw: 153.7
<br>18:28:01 checking zb system running ok
<br>18:28:03 polled inv 0 ts: 119 pw: 106.6
<br>18:30:01 checking zb system running ok
<br>18:30:03 polled inv 0 ts: 120 pw: 62.9
<br>18:32:01 checking zb system running ok
<br>18:32:03 polled inv 0 inverter restarted <================
<br>18:32:03 polled inv 0 ts: 105 pw: 71.0
<br>18:34:02 checking zb system running ok
<br>18:34:04 polled inv 0 ts: 121 pw: 108.2
<br>18:36:01 checking zb system running ok
<br>18:36:03 polled inv 0 ts: 119 pw: 100.0
<br>18:38:01 checking zb system running ok
<br>18:38:03 polled inv 0 ts: 120 pw: 76.2
```

rasor89, 2022-06-24:
I can't further investigate at the moment. I have issues to connect with my wifi and don't know the reasons. Maybe my wifi stick is damaged :(

closed 2022-06-24 by patience4711