patience4711/RPI-APS-inverters #51: inverterPoll.cgi in new update ecu_swupdate_14-06-2022 (not work for me)

b31k, 2022-06-21:
hello good morning the new update did not work for me, the inverters stopped communicating, I went back to the inverterPoll.cgi version of the RPI-ECU-v2_0 image, and the problem was solved.

thank you very much for your work
best regards

b31k, 2022-06-21:
> @b31k<https://github.com/b31k> Sorry to hear that. Too bad you didn't provide more details, now it's hard to debug.
> So could you tell me what inverters do you have?

yes
we have 3 APSystem QS1 1200w

patience4711, 2022-06-21:
You say the new update, could you be specific on that, what exactly did you do?
Maybe it has something to do with the settings, other ecu-id maybe?

b31k, 2022-06-21:
I was simply on RPI-ECU-v2_6 and then I downloaded RPI-ECU-v2_7, installed it and then RPI-ECU-v2_8 and updated again.
Then I saw that the inverters were not communicating, so I investigated the modified files in these updates and I saw that one of them was the inverterPoll.cgi, I replaced it with the image I had of RPI-ECU 2.0 and it started to work.
If you want more specific information, let me know and I'll send it to you.

thank you very much for your work

patience4711, 2022-06-21:
My version 2_8 works perfectly with the yc600 i have so i cannot think of any reason why it wouldn't work with the qs1, i didn't touch that part of the script. Only some changes in the ds3 part. 
You could provide more info if you run version 2_8 and do a manual poll via the console. This gives a lot of debug info so that the error can be found easely.

b31k, 2022-06-23:
Hi, today the inverters did not wake up to communicate with zigbee, maybe this is the fault and not the version.(sorry)
Could it be that the amount of sun (watts) make it communicate or not?

maybe what happened was that when updating there was no sun and that's why it didn't communicate?

this is today's log, you can see how it didn't work first thing in the morning and when the sun came out it started to work.
at 11:00
[image]

2022-06-23 01:00:00 running midNight.cgi
00:00:04 midNight.cgi moved energy total into db invEnergy
midNight: removed data files
sun.py : calculated sunrise / sunset
00:04:03 checking zb system running ok
06:30:06 woke up from sleep, goodmorning!
going to perform the morning ritual
invdataNull.cgi: wrote a new /ramdisk/invData0 with null values
invdataNull.cgi: wrote a new /ramdisk/invData1 with null values
invdataNull.cgi: wrote a new /ramdisk/invData2 with null values
Wrote 0 in database invData,
1st poll in 5 min, now exit
06:35:09 polled inv 0
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData0
06:35:12 polled inv 1
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData1
06:35:14 polled inv 2
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData2
06:40:08 polled inv 0
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData0
06:40:10 polled inv 1
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData1
06:40:13 polled inv 2
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData2
06:45:08 polled inv 0
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData0
06:45:10 polled inv 1
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData1
06:45:13 polled inv 2
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData2
06:50:10 polled inv 0
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData0
06:50:12 polled inv 1
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData1
06:50:15 polled inv 2
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData2
06:55:08 polled inv 0
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData0
06:55:10 polled inv 1
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData1
06:55:13 polled inv 2
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData2
07:00:05 checking zb system failed, no answer on serial
07:00:11 polled inv 0
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData0
[... 449 lines cut ...]
12:15:11 polled inv 2 ts: 300 pw: 724.9
12:20:08 polled inv 0 ts: 301 pw: 586.8
12:20:10 polled inv 1 ts: 301 pw: 687.1
12:20:12 polled inv 2 ts: 301 pw: 679.1
12:25:07 polled inv 0 ts: 299 pw: 644.9
12:25:09 polled inv 1 ts: 299 pw: 758.0
12:25:11 polled inv 2 ts: 299 pw: 754.7
12:30:07 polled inv 0 ts: 300 pw: 650.4
12:30:09 polled inv 1 ts: 300 pw: 753.6
12:30:11 polled inv 2 ts: 300 pw: 745.2
12:35:09 polled inv 0 ts: 302 pw: 782.9
12:35:10 polled inv 1 ts: 301 pw: 891.3
12:35:12 polled inv 2 ts: 302 pw: 889.5
12:40:07 polled inv 0 ts: 298 pw: 888.1
12:40:09 polled inv 1 ts: 299 pw: 1016.5
12:40:11 polled inv 2 ts: 298 pw: 1016.6
12:45:07 polled inv 0 ts: 300 pw: 767.2
12:45:09 polled inv 1 ts: 300 pw: 871.0
12:45:11 polled inv 2 ts: 300 pw: 864.3
12:50:07 polled inv 0 ts: 300 pw: 866.4
12:50:09 polled inv 1 ts: 300 pw: 970.2
12:50:11 polled inv 2 ts: 300 pw: 967.1
12:55:07 polled inv 0 ts: 300 pw: 756.0
12:55:09 polled inv 1 ts: 300 pw: 878.9
12:55:11 polled inv 2 ts: 300 pw: 876.6

patience4711, 2022-06-23:
If there is no answer on serial, only a fault in the wiring can be the cause. Check the wiring and soldering for a bad connection. Maybe try another powersupply. 

To improve the issues as a knowledge database could you please edit the title of the issue. The update is not the cause here.