patience4711/RPI-APS-inverters #52: Sources without burning RPi Image

fellek, 2022-06-22:
Before opening an issue please consider this: This project works with the described hardware as fas as i could test it. So some feedback is always welcome. Please state what hardware you use, raspberry platform, kind of inverter(s) you have and how many.

patience4711, 2022-06-22:
It's the easiest and least labor intensive way for me. It is not a matter of just sharing some scripts you know.

fellek, 2022-06-22:
Thanks for the straigth answer. I wonder wether there is a specific difference between the esp /raspi programs and if you have plans to share them. In my specific setup I would like to use my running raspi for polling my inverters (of course there are two DS3-S, they are not supported by version 2.06)

patience4711, 2022-06-23:
The difference is the programming languages, in the RPI we have C++, Perl, PHP, Javascript, Bash, python, maybe i forgot one. In the ESP only Arduino C.
I think you would like a package that installs this system on an existing Raspberry. No plans for that, too complicated for me.