patience4711/RPI-APS-inverters #53: having a new variable for totalen

nlienard, 2022-06-23:
currently totalen is reseted every day and due to that, integration in Home Assistant is having troubles.

it is possible to have a variable totalen2 (or whatever) which is a total which always increments for ever ?

thanks

patience4711, 2022-06-23:
Not possible, when this value is written to a file we would wear out our sdcard, when saved in ramdisk it would disappear at a powercycle.

rasor89, 2022-06-24:
would it be an possible option to write down the values in an external influxdb?

nlienard, 2022-06-24:
another question: how do you create totalen currently ? i don't see any doc explaining the raw output of the invecter.
I thought it sends the totalen as raw, as a total always incrementing but if i understand, when inverter restarts, it doesn't keep in its internal memory the previous value and began back to 0 ?

patience4711, 2022-06-24:
totalen is the value polled from the inverter. Will be reset after a inverter restart. Please read the wiki subject inverters

nlienard, 2022-06-24:
finaly it is not an issue if inverter restarts and put totalen to 0 because home assistant is smart and manages to correct it without any action

[image]

closed 2022-06-24 by nlienard