patience4711/RPI-APS-inverters #59: Version 3.3 no Totals / stats

rasor89, 2022-08-06:
Hello @patience4711 

after the update to version 3.3 I have no link for the stats (headbar is missing!) Also there is no Totals available. See below picture. 
Because I wasn't sure, that the update reasoned this problem, I flashed the system complety new (3.3 img) and restored the backup. But the failure is still available. 
[image]

patience4711, 2022-08-09:
Sorry to hear that, When you click the button that says DS3 you should get the details page with buttons to the statistics and the charts.
So you say when there are 2 panels everything works like it should. With one panel there are no figures in the main page. I have to investigate this.

patience4711, 2022-08-15:
No thanks...

closed 2022-08-15 by patience4711

rasor89, 2022-08-17:
@patience4711 sorry for not replying, but didn't found time the last days for testing the new version. I went back to the 2.9 to have a working system. Maybe I will find some time in the next days.