patience4711/RPI-APS-inverters #60: What does "inMessage = FE0164020067" mean?

Cryss67, 2022-08-17:
Hello!

I try to pair my inverter (DS3-S) on my Raspberry 3b using your v3.3 image.
Do you need any more information from me?

I've successfully copied the firmware to the Zigbee module, but on pairing I get this message and I can't figure out it's meaning.

```

running script inverterPair.pl...
argument is 0 
sudo /usr/lib/cgi-bin/ecu/inverterPair.cgi 0 
Content-type:text/html
running inverterPair.cgi

inverterpair running with arg0
wiringPi started 
opened port serial0 
fileToread = /var/www/ecu_data/ecuProperties
json from file = { 'idinvers': "80971B01A3D8", 'id': "D8A3011B9780" } 
idinvers from file = 80971B01A3D8 
ecushort = A3D8 
fileToread = /var/www/ecu_data/inverters/invProperties0
json from file = { 'panels': [ 1, 1, 0, 0 ], 'idx': "", 'id': "unpaired", 'name': "Inverter 1", 'type': 2, 'serial': 702000014595 } serial from file = 702000014595 
trying to pair inverter0 
send cmd 0 2524020FFFFFFFFFFFFFFFFF14FFFF140D0200000F1100702000014595FFFF10FFFF80971B01A3D8FA 
inMessage = FE0164020067 
send cmd 1 1A24020FFFFFFFFFFFFFFFFF14FFFF140C0201000F0600702000014595B4 
inMessage = FE0164020067 
no pairing answer, returning... 
proceeding with the next cmd 
send cmd 2 2524020FFFFFFFFFFFFFFFFF14FFFF140F0102000F1100702000014595A3D810FFFF80971B01A3D882 
decode inMessage = FE0164020067 
no pairing answer, returning... 
no pairing answer, returning... 
proceeding with the next cmd 
send cmd 3 1A24020FFFFFFFFFFFFFFFFF14FFFF14010103000F060080971B01A3D84F
decode inMessage = FE0164020067 
no pairing answer, returning... 
all pair cmds done
info = 5 
pairing failed, try again... 
inverterPair ready 
pair command processed 
HTTP:1/1 200 OK
```

Healthcheck output:

```
running script checkCoordinator.pl...
calling checkCoordinator.cgi command to run = sudo /usr/lib/cgi-bin/ecu/checkCoordinator.cgi
Content-type:text/html
running checkCoordinator.cgi

wiringPi started 
opened port serial0 
command = 00270027 
answer = FE0E670000FFFF80971B01A3D8000007090011 rc=19 
fileToread = /var/www/ecu_data/ecuProperties
idinvers from file = 80971B01A3D8 
coordinator is running ok 
toLog = 
18:31:54 checking zb system running ok 
logCmd = echo "
18:31:54 checking zb system running ok" >> /ramdisk/ecu_log.txt 
command runned
HTTP:1/1 200 OK
```

Cryss67, 2022-08-18:
So it SEEMS that it was an issue of distance. I wasn't able to pair the inverter with the Zigbee module at 1 meter, but now it worked 50 cm away. Is this possible?
What's the point of a Wifi module that needs to be that close?
Is it possible that the radio power is set is too low and needs to be adjusted?

closed 2022-08-18 by Cryss67

rasor89, 2022-08-22:
@Cryss67 could you increase the range? Can you test the max. possible? I'm still looking for options to increase the signal strengh, because at the moment I can only realize small distances between my RPI and the converter.

Cryss67, 2022-08-22:
Currently, the Zigbee module is about 5 meters away from the inverter with two walls  (I guess Y-Tong) in between. I‘ll try to test the range although I need to find a mobile solution for the raspi as I live on the upper floors of a rented apartment with no power supply

Cryss67, 2022-08-22:
BTW, what other features have you planned?