patience4711/RPI-APS-inverters #61: What are these inverter spikes?

Cryss67, 2022-08-22:
is there a way to „normalize“ these, I guess wrong values?

[image]

patience4711, 2022-08-24:
What systemversion/inverter? They are most likely the result of a disturbed polling/zigbee communication. You can check that in the journal and in the timeline of the chart. The system ignores unrealistic values so in the end the total energy value should be oke.
If not you can change the value for that day in the energy database.

Cryss67, 2022-08-26:
Software version: RPI-ECU-v3_5
The inverter is a DS3-S.

Thanks for your help!

closed 2022-08-31 by patience4711