patience4711/RPI-APS-inverters #63: Possible to pair rpiecu in addition to ECU-B?

AB-274, 2022-08-28:
I'm going to build this up, thanks for all the stuff in advance. One question though: Is it possible to pair the rpiecu in addition to an existing paired ECU-B (using a DS3-S inverter)? 
Or can an inverter only be paired to one single device?
Thanks!

patience4711, 2022-08-31:
I am able to poll the inverter with 2 different rpi-ecu's but the are both using the same ecu-ID. 
You can only pair to one ecu but if the second one has the same proporties (ecu-id and inverter short id) it should work

AB-274, 2022-09-01:
Thanks for the input! I have the hardware ready so I could give it a try. The ecu-ID of the rpiecu is the 12 char one that can be set in the settings (so I would set this one to be the same like the original Ecu-B that I have up and running)?
What is the Inverter short ID exactly - do you have an idea how to look up that ID for the existing ECU-B communication?

And in general, would it also be possible to kind of sniff the polls started by the original Ecu-B without starting polls from the rpiecu itself? (Thinking of reducing the risk of collisions with two devices polling one inverter. Not sure if this is an issue anyway...)

Background is that I would really like to use rpiecu  to grab the data and use it in my smart home environment using mqtt. At the same time, the original ecu-b is running fine and I would like to keep it up and running because of the warranty extension and possible firmware updates.

Hansiart2, 2022-09-10:
The rpiecu polls by sending a datarequest and than listens for the answer. So it does dot receive answers to requests from other devices. If you pair your inverter you can see the short id in the id field, something like 0xAB12. Than pair back to to the ecu-b.

closed 2022-09-19 by patience4711