patience4711/RPI-APS-inverters #67: Crash of my RPI - how to recover ?

nlienard, 2022-09-09:
Hi, 

I'm doing to reinstall a fresh image OS to my RPI3 and need to know which files are neeful to keep the current configuration i got to have the inverters config and so on.

Context:

After a storm, my rpi3 has been cut from electricity 3 times in 5mn and now is not booting anymore 

```
Kernel panic-not syncing: VFS: unable to mount root fs on unknown- block
```

I can see reach the file of the SD card from my Odroid (I hope, sometimes it hangs), so instead of doing a full backup (it frozes after a while), i would like to focus on specific useful files which contains the inverters config and so on.

```
Sep  9 14:21:23 odroid-nas kernel: [107315.099065] usb 1-2: new high-speed USB device number 2 using xhci-hcd
Sep  9 14:21:23 odroid-nas kernel: [107315.247972] usb 1-2: New USB device found, idVendor=058f, idProduct=6366, bcdDevice= 1.00
Sep  9 14:21:23 odroid-nas kernel: [107315.247990] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Sep  9 14:21:23 odroid-nas kernel: [107315.247995] usb 1-2: Product: Mass Storage Device
Sep  9 14:21:23 odroid-nas kernel: [107315.247999] usb 1-2: Manufacturer: Generic
Sep  9 14:21:23 odroid-nas kernel: [107315.248003] usb 1-2: SerialNumber: 058F63666433
Sep  9 14:21:23 odroid-nas kernel: [107315.250091] usb-storage 1-2:1.0: USB Mass Storage device detected
Sep  9 14:21:23 odroid-nas kernel: [107315.250729] scsi host2: usb-storage 1-2:1.0
Sep  9 14:21:23 odroid-nas kernel: [107315.387396] usbcore: registered new interface driver uas
Sep  9 14:21:24 odroid-nas kernel: [107316.391512] scsi 2:0:0:0: Direct-Access     Multiple Card  Reader     1.00 PQ: 0 ANSI: 0
Sep  9 14:21:24 odroid-nas kernel: [107316.391912] sd 2:0:0:0: Attached scsi generic sg2 type 0
Sep  9 14:21:25 odroid-nas kernel: [107317.590375] sd 2:0:0:0: [sdc] 31481856 512-byte logical blocks: (16.1 GB/15.0 GiB)
Sep  9 14:21:25 odroid-nas kernel: [107317.590765] sd 2:0:0:0: [sdc] Write Protect is on <===== VERY BAD !!!!!!!!!!!!!!
Sep  9 14:21:25 odroid-nas kernel: [107317.596307]  sdc: sdc1 sdc2
Sep  9 14:21:25 odroid-nas kernel: [107317.598284] sd 2:0:0:0: [sdc] Attached SCSI removable disk
```

```
root@odroid-nas:~# mount /dev/sdc2 /mnt
mount: /mnt: cannot mount /dev/sdc2 read-only.
```

```
root@odroid-nas:~# hdparm -r0 /dev/sdc

/dev/sdc:
 setting readonly to 0 (off)
 readonly      =  1 (on)
```

```
root@odroid-nas:~# mount -o ro,noload /dev/sdc2 /mnt
root@odroid-nas:~# cd /mnt
root@odroid-nas:/mnt# ls
'$'   bin   boot   boot1   check.txt   dev   etc   home   lib   lost+found   media   mnt   opt   proc   ramdisk   root   run   sambamap   sbin   snap   srv   stuff   sys   testdir   titels_temp.txt   tmp   usr   var
```

Where are the important files (inverters config etc) to copy to new sd card ?

thanks

nlienard, 2022-09-09:
For now, i copied that:

```
root@odroid-nas:/mnt/var/www/ecu_data# cp -rp inverters /data/stuff/DS3L/backup
root@odroid-nas:/mnt/var/www/ecu_data#
```

I would be happy if you could tell me the other files to copy

Hansiart2, 2022-09-10:
The directory you coppied is the right one. The ones you cant copy have a different owner. Restoring your database is something else. Maybe the documentation of influx can help.
I think however that your fs can be repaired. Linux has a program for that.

nlienard, 2022-09-10:
thanks.

For the SD card, it looks dead

```
[108945.828154] sd 2:0:0:0: [sdc] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x00 driverbyte=DRIVER_OK cmd_age=4s
[108945.828173] sd 2:0:0:0: [sdc] tag#0 Sense Key : 0x2 [current]
[108945.828179] sd 2:0:0:0: [sdc] tag#0 <<vendor>>ASC=0xff <<vendor>>ASCQ=0xff
[108945.828186] sd 2:0:0:0: [sdc] tag#0 CDB: opcode=0x28 28 00 00 6d 73 e8 00 00 08 00
[108945.828191] blk_update_request: I/O error, dev sdc, sector 7173096 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 0
[108951.045671] sd 2:0:0:0: [sdc] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x00 driverbyte=DRIVER_OK cmd_age=4s
[108951.045688] sd 2:0:0:0: [sdc] tag#0 Sense Key : 0x2 [current]
[108951.045695] sd 2:0:0:0: [sdc] tag#0 <<vendor>>ASC=0xff <<vendor>>ASCQ=0xff
[108951.045703] sd 2:0:0:0: [sdc] tag#0 CDB: opcode=0x28 28 00 00 6d 73 e8 00 00 08 00
[108951.045708] blk_update_request: I/O error, dev sdc, sector 7173096 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[108956.278828] sd 2:0:0:0: [sdc] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x00 driverbyte=DRIVER_OK cmd_age=4s
[108956.278846] sd 2:0:0:0: [sdc] tag#0 Sense Key : 0x2 [current]
[108956.278852] sd 2:0:0:0: [sdc] tag#0 <<vendor>>ASC=0xff <<vendor>>ASCQ=0xff
[108956.278859] sd 2:0:0:0: [sdc] tag#0 CDB: opcode=0x28 28 00 01 e0 5f 80 00 00 08 00
[108956.278864] blk_update_request: I/O error, dev sdc, sector 31481728 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 0
```

nlienard, 2022-09-10:
```
[192150.467816] EXT4-fs (sdc2): INFO: recovery required on readonly filesystem
[192150.467830] EXT4-fs (sdc2): write access unavailable, cannot proceed (try mounting with noload)
root@odroid-nas:~#  mount  -o noload /dev/sdc2 /mnt
mount: /mnt: WARNING: source write-protected, mounted read-only.
root@odroid-nas:~#
```

nlienard, 2022-09-10:
Today i bought a new SD card and reinstall ECU RPI on it but the latest version 3.3

Then i copied the ecu_data files as well from my previous SD CARD

and when checking coordinator i got this error

```
<title>RPI ECU PROGAM</title>
</head>
<body>
<h4>running checkCoordinator.cgi</h4>
wiringPi started
<br>
opened port serial0
<br>
command = 00270027
<br>
answer = 7F84782E rc=4
<br>
fileToread = /var/www/ecu_data/ecuProperties<br>
idinvers from file = 80971B01B3D6
<br>
an unknown error occurred
<br>
toLog = <br>19:26:20 checking zb system failed, an unknown error occurred
<br>
logCmd = echo "<br>19:26:20 checking zb system failed, an unknown error occurred" >> /ramdisk/ecu_log.txt
<br>
```

what does that  mean ?

thanks

nlienard, 2022-09-10:
after running  " /usr/lib/cgi-bin/ecu/startCoordinator.pl normal=1" it looks better:

```

<html>
<head>
<title>RPI ECU PROGAM</title>
</head>
<body>
<h4>running checkCoordinator.cgi</h4>
wiringPi started
<br>
opened port serial0
<br>
command = 00270027
<br>
answer = FE0E670000FFFF80971B01B3D600000709000F rc=19
<br>
fileToread = /var/www/ecu_data/ecuProperties<br>
idinvers from file = 80971B01B3D6
<br>
coordinator is running ok
<br>
toLog = <br>19:31:09 checking zb system running ok
<br>
logCmd = echo "<br>19:31:09 checking zb system running ok" >> /ramdisk/ecu_log.txt
```

but polling are still empty

nlienard, 2022-09-11:
OK this morning it looks fine !

[image]

nlienard, 2022-09-11:
i closed this issue, thanks

closed 2022-09-11 by nlienard