patience4711/RPI-APS-inverters #70: Firmware version of the inverter ?

nlienard, 2022-09-27:
Hi

Is it possible to get the Firmware version of the inverter (in my case DS3L). ?

https://forum-photovoltaique.fr/download/file.php?id=9943&mode=view

Thanks !

patience4711, 2022-09-27:
If you can find the corresponding zigbee command the answer maybe yes. You can issue that command (if it exists) via the console. Maybe you can ask one of the guys [here](https://github.com/Koenkk/zigbee2mqtt/issues/4221)<br>I cannot find it. 
Good luck..

patience4711, 2022-10-07:
Looks like someone is able to get the firmware version, check this:
https://gitlab.com/moreroid/apsystems/-/blob/main/firmware/apsystems-firmware-download.sh

nlienard, 2022-10-07:
yes, the link about is this one i guess
https://gitlab.com/moreroid/apsystems/-/blob/main/zigbee/apsystems-zigbee-ds3-poll.txt
the one you mentionned is about firmware update ?

Cryss67, 2022-10-08:
Great!

BTW, are the changelogs for firmware changes (DS3-S) to be found somewhere or have I overlooked them?

nlienard, 2022-10-19:
anyone is able to interpet the previous post with hexa code to have a proper zigbee command to send to the inverter for having Firmware version ? thanks

defaliz, 2022-11-15:
Bonsoir Nicolas
j'ai de gros problèmes pour apairer zigbee et mon APS ds3L
à quelle distance faut-il être du micro onduleur ? j'utilise un raspberry
merci pour votre réponse
Alain

nlienard, 2023-01-23:
@defaliz j'ai eu pas mal de souci aussi avec la portée, je dirais qu'il faut être à moins de 5m si y a trop d'obstacles.
j'ai acheté une antenne avec une rallonge filaire pour rapprocher l'antenne le plus possible de mon DS3L. c'est désormais près de la fenêtre et le signal est mieux et j'ai plus de pb.

defaliz, 2023-01-23:
Merci Nicolas
j'ai remarqué également qu'avec un mur en brique entre les 2, 5m c'est le max..
tu as mis ton antenne sur un raspberry pi ou zero ?

nlienard, 2023-01-23:
sur un rpi3B+

defaliz, 2023-01-23:
merci tu as un lien qui montrerai le montage ? j'ai rien trouvé..
merci encore

defaliz, 2023-01-23:
Merci beaucoup, je regarde ça
bonne soirée

closed 2023-02-16 by patience4711