patience4711/RPI-APS-inverters #71: New DS3(L) version ?

nlienard, 2022-09-29:
When wanting to buy in FRANCE new DS3L inverter, the sales told me the new version from AUGUST is having big changes : the inverter DS3L won't work OUT OF THE BOX, it would require to connect at first to ECU in order to adapt the FREQUENCY (50Hz or 60Hz) and tension (220V or 110V) according to the country where it is used.

Anybody heard about that ? i'm a bit afraid to buy a new DS3L if i can't use it :(

thanks

patience4711, 2022-09-29:
That would imply that the inverter is useless without an aps ECU. Hard to believe but you should be able to read that in the manual of the inverter. This is not an issue in the scope of the esp-ecu software.

nlienard, 2022-10-03:
I got response from APS saying it still work without ECU but warranty is only 2 years without ECU.
They say ECU is needful for firmware update but maybe soon, with your code, we'll be able to update inverter firmware one day !

patience4711, 2022-10-04:
Oke thanks for sharing this. 
I don't think that we are able to update firmware on the inverters on short notice. If someone with an APS ECU can sniff the zigbee trafic and is willing to experiment (which is obviously not without risks), it would be possible.

closed 2022-10-04 by patience4711