patience4711/RPI-APS-inverters #73: Would ZigBee USB Stick with coordinator firmware work?

mi-hol, 2022-10-18:
For ease of setup I'd prefer to buy a ZigBee USB Stick with coordinator firmware.

Are there perhaps already experiences with specific models (good or bad)?
I intend to try with "SONOFF Zigbee Gateway USB Dongle"  from a German seller on ebay. 
"Modell: MM SmartHome ZIGBEE CC2652P USB COORDINATOR"

In case it was never tried before I'd be happy to be the guinea pig to test such a setup, assuming I could get a helping hand :)

Hardware:
- raspberry platform: RPI 1
- kind of inverter: single DS3S

patience4711, 2022-10-18:
That won't work for several reasons. F.o.a. you need to run the firmware special for this project (the wiki points this out) and second the zigbee communicantion does not run on usb port.

mi-hol, 2022-10-19:
> need to run the zigbee firmware special for this project (the wiki points this out)

Thanks for clarification, is this meant by "These modules have to be flashed with the firmware developped for this purpose by @kadszol (included in the download package)." ?

As I didn't find user kadszol in GitHub or elsewhere therefore I wonder, is the source code for this firmware published somewhere?

So far my only source of information is this repo, are there other sources (i.e. forum)?

patience4711, 2022-10-19:
You can find more [here](https://github.com/Koenkk/zigbee2mqtt/issues/4221) where @kadzsol shared this firmware.

closed 2022-10-19 by patience4711