patience4711/RPI-APS-inverters #77: DS3-H 3ª generation(2022),  inverter is not working?

b31k, 2022-11-18:
Hello good morning
I have added 2 new ds3-h (2022) inverters to my installation, but unfortunately I don't receive any data in the poll, however the pairing is doing very well.
note:**The QS1 that I have do work properly.**

INVERTER 1

running script inverterPoll.pl...
argument is 1
sudo /usr/lib/cgi-bin/ecu/inverterPoll.cgi 1
Content-type:text/html
running inverterPoll.cgi
inverterpoll running with arg 1
invSerial from file = 705000002462
invType = 2
idx = 801
invIdrev = 63E4
idShort = xxxx
idinvers from file = 80971B01A3D8
poll cmd = 1D240163E41414060001000F1380971B01A3D8FBFB06BB000000000000C1FEFEAE
no answer
all cmds sent
returned by sendCommands() 4098
toLog =
15:20:42 polled inv 1
logCmd = echo "
15:20:42 polled inv 1 " >> /ramdisk/ecu_log.txt
Content-type:text/html
running invdataNull.cgi
echo "{ 'dcc': [ 0, 0, 0, 0 ], 'idx': 0, 'totalen': 0, 'freq': 0, 'pwr': [ 0, 0, 0, 0 ], 'time': 0, 'acv': 0, 'totalpw': 0, 'heath': 0, 'dcv': [ 0, 0, 0, 0 ], 'en': [ 0, 0, 0, 0 ] }" > "/ramdisk/invData1"
sedCmd = sed -i 's|['\'']|'\"'|g' /ramdisk/invData1
invdataNull.cgi: resetted existing invData
invdataNull.cgi exit with code 0
inverterpoll exit with code 0
command processed
HTTP:1/1 200 OK

INVERTER 2
running script inverterPoll.pl...
argument is 3
sudo /usr/lib/cgi-bin/ecu/inverterPoll.cgi 3
Content-type:text/html
running inverterPoll.cgi
inverterpoll running with arg 3
invSerial from file = 705000003659
invType = 2
idx = 803
invIdrev = 4D98
idShort = xxxx
idinvers from file = 80971B01A3D8
poll cmd = 1D24014D981414060001000F1380971B01A3D8FBFB06BB000000000000C1FEFEFC
no answer
all cmds sent
returned by sendCommands() 4098
toLog =
15:21:25 polled inv 3
logCmd = echo "
15:21:25 polled inv 3 " >> /ramdisk/ecu_log.txt
Content-type:text/html
running invdataNull.cgi
echo "{ 'idx': 0, 'totalen': 0, 'freq': 0, 'heath': 0, 'dcv': [ 0, 0, 0, 0 ], 'time': 0, 'pwr': [ 0, 0, 0, 0 ], 'acv': 0, 'en': [ 0, 0, 0, 0 ], 'dcc': [ 0, 0, 0, 0 ], 'totalpw': 0 }" > "/ramdisk/invData3"
sedCmd = sed -i 's|['\'']|'\"'|g' /ramdisk/invData3
invdataNull.cgi: resetted existing invData
invdataNull.cgi exit with code 0
inverterpoll exit with code 0

patience4711, 2022-11-21:
So what sw version do you use?

b31k, 2022-11-21:
RPI ECU INFORMATION
software version        RPI-ECU-v3_6

patience4711, 2022-11-23:
Sorry to hear that. I have never tested this inverter so i am afraid that i can't help you with this. I gues someone has to go to the whole process of sniffing the zigbee trafic between an official ECU and the inverter to find out what is going on. 

However, when it pairs it should answer to the poll commands. There are more people having troubles with the polling sometimes, this seems often a range problem. 

I hope that someone else can confirm that this inverter does or does not work with the rpi-ecu.

leola, 2022-11-25:
Hello

Not sure if linked but, I am runing RPI-ECU-v3_3
I have a bran new DS3-L and I have similar issue and logs : 

running script inverterPoll.pl...
argument is 0
sudo /usr/lib/cgi-bin/ecu/inverterPoll.cgi 0
Content-type:text/html
running inverterPoll.cgi
inverterpoll running with arg 0
invSerial from file = 703000327796
invType = 2
idx = 1
invIdrev = 9C75
idShort = xxxx
idinvers from file = 80971B01B3D6
poll cmd = 1D24019C751414060001000F1380971B01B3D6FBFB06BB000000000000C1FEFEDE
no answer
all cmds sent
returned by sendCommands() 4098
toLog =
14:29:35 polled inv 0
logCmd = echo "
14:29:35 polled inv 0 " >> /ramdisk/ecu_log.txt
Content-type:text/html
running invdataNull.cgi
echo "{ 'acv': 0, 'idx': 0, 'time': 0, 'pwr': [ 0, 0, 0, 0 ], 'dcc': [ 0, 0, 0, 0 ], 'en': [ 0, 0, 0, 0 ], 'totalen': 0, 'freq': 0, 'totalpw': 0, 'dcv': [ 0, 0, 0, 0 ], 'heath': 0 }" > "/ramdisk/invData0"
sedCmd = sed -i 's|['\'']|'\"'|g' /ramdisk/invData0
invdataNull.cgi: resetted existing invData
invdataNull.cgi exit with code 0
inverterpoll exit with code 0
command processed
HTTP:1/1 200 OK

I managed to pair it ok but nothing else works

nlienard, 2023-01-10:
Hi
i see your serial of DS3L is 703000327796.
you are in the BAD range about deffected DS3L. Mine was also in this range, and i contacted the resseler to proceed to a replacement.
For info, the bad SERIAL are: from 703000195xxx to  7030003289xx.

you can see the info on many site: https://www.quesolar.com/apsystems-storing-ds3-en-ds3-l-oktober-november-2022/
use google translate.

b31k, 2023-01-11:
Hi nicolas my micro inverters, has 705000003659 and
705000002462, other are two QS1

nlienard, 2023-01-12:
> Hi nicolas my micro inverters, has 705000003659 and 705000002462, other are two QS1
> […](#)

should be good, but best is to call your resseler if you have issue or any doubt.

leola, 2023-02-07:
> should be good, but best is to call your resseler if you have issue or any doubt. My response was for @leola who got issue and has a serial in the bad range.

Thanks Nicolas for the info on deffected DS3L inverters , I had it replaced by APSytem, 
No idea what was wrong in this series but they send a new one right away.

Furthermore, it does not fix the communication issue using RPI-RCU and my 2 DS3L, the pooled data is empty : 

running script inverterPoll.pl...
argument is 1
sudo /usr/lib/cgi-bin/ecu/inverterPoll.cgi 1
Content-type:text/html
running inverterPoll.cgi
inverterpoll running with arg 1
invSerial from file = 703000570026
invType = 2
idx = 2
invIdrev = B052
idShort = xxxx
idinvers from file = 80971B01B3D6
poll cmd = 1D2401B0521414060001000F1380971B01B3D6FBFB06BB000000000000C1FEFED5
answer = FE0164010064FE034480001401D2FE0345C4B0520060FE0345C4B0520060
messageToDecode = FE0164010064FE034480001401D2FE0345C4B0520060FE0345C4B0520060
no AF_INCOMING_MSG
all cmds sent
returned by sendCommands() 13
toLog =
14:51:44 polled inv 1 error 10 AF_INCOMING_MSG fail
logCmd = echo "
14:51:44 polled inv 1 error 10 AF_INCOMING_MSG fail" >> /ramdisk/ecu_log.txt
Content-type:text/html
running invdataNull.cgi
echo "{ 'heath': 0, 'dcv': [ 0, 0, 0, 0 ], 'freq': 0, 'totalen': 0, 'acv': 0, 'en': [ 0, 0, 0, 0 ], 'dcc': [ 0, 0, 0, 0 ], 'idx': 0, 'totalpw': 0, 'pwr': [ 0, 0, 0, 0 ], 'time': 0 }" > "/ramdisk/invData1"
sedCmd = sed -i 's|['\'']|'\"'|g' /ramdisk/invData1
invdataNull.cgi: resetted existing invData
invdataNull.cgi exit with code 0
inverterpoll exit with code 0
command processed
HTTP:1/1 200 OK 

Anyone keen on sniffing the communication using an official ECU ?

Thanks

closed 2023-02-16 by patience4711