patience4711/RPI-APS-inverters #79: change ip dhcp to static.. ??

defaliz, 2022-12-28:
hi,
it seems changing ip dhcp to static with the web interface doesn't work..menu, ip config..
i've try to change in /etc/network/interfaces but the same.. not changing to static ip
how can i do this ?
regards

bjoerg560, 2022-12-28:
I have the same problem.

patience4711, 2023-01-03:
The script /usr/lib/cgi-bin/ecu/wifi/wifi_static.pl changes the file /etc/network/interfaces

The file etc/network/interfaces should look like:

**source-directory /etc/network/interfaces.d
iface wlan0 inet static
address <IP>
netmask 255.255.255.0
network 192.168.0.0
gateway 192.168.0.1 
broadcast 192.168.0.255**

**nwpa-conf /etc/wpa_supplicant/wpa_supplicant.conf**

edit  /var/www/ecu_data/wifistatus.txt
echo static > /var/www/ecu_data/wifistatus.txt

check wpa_supplicant.conf for the right credentials

defaliz, 2023-01-03:
Thanks a lot..
i look for this file and back to tell you...
have an Happy New Year 2023 !

nlienard, 2023-01-10:
oh i didn't know about this file, thanks

i put the network file immuable with chattr because i was fedup it reboots.

root@192.168.1.180 ~/MAJ# lsattr /etc/network/interfaces
----i---------e---- /etc/network/interfaces

closed 2023-01-21 by patience4711