patience4711/RPI-APS-inverters #81: mqtt json issue when having a second inverter

nlienard, 2023-01-10:
With 1 inverter, all was good, i got a json with pw0 and pw1 value and so on.

Today i added a second inverter and instead of having pw2 and pw3 in same json,, it generates a second json with again pw0 and pw1 (and of course serial to differenciate).

I'm using Home Assistant and i m stuck because i can't define a new sensor for pw2 as pw2 doesn't exist, it is pw0/pw1 again and it is overring my existing value; it is a total mess.

isn't it possible to choose a MQTT FORMAT with  pw0 pw1 pw2 pw3 in SAME json ?

i'm running latest version 3.6

patience4711, 2023-01-21:
The mqtt messages are per inverter. Each has 1-4 pv units so the values are pw0-pw3 etc. Never knew that this could cause troubles. I think that this is even to Home Assistent logical. The json contains the serialnr, with this you can distinguish the values.

nlienard, 2023-01-23:
finally, i fixed my issue.

```
mqtt:
  sensor:
    - name: "pw0"
      state_topic: "aps/in"
      unit_of_measurement: "W"
      value_template: >
        {% if value_json is defined and value_json.serial == '703000069723' %}
          {{ value_json.pw0 }}
        {% else %}
          {{ states('sensor.pw0') }}
        {% endif %}
      payload_available: "online"
      payload_not_available: "offline"
      unique_id: “pw0”
      state_class: total
      device_class: power
```

closed 2023-01-23 by nlienard