patience4711/RPI-APS-inverters #82: mosquitto retain flag

nlienard, 2023-01-10:
Is it possible to give us the ability to add the retain flag when sending mqtt msg through mosquitto ?
right now, it is done by CGI  /usr/lib/cgi-bin/ecu/sendMqtt.cpp

 string systemCmd="mosquitto_pub -h " + mqtt_adr + " -p " + mqtt_port + " -t " + mqtt_out + " -m " + toMqtt ;

retain flag is useful for home assistant, to tell him to keep the last value.

thanks

patience4711, 2023-01-23:
I added a checkbox in the mqt configuration page to use the retainflag or not. When i tested this i'l provide the changed webpage and scripts that you can install yourself. In the next release this will be incorporated.

patience4711, 2023-01-24:
If you download this zip : https://github.com/patience4711/RPI-APS-inverters/blob/main/retain.zip
and replace the files on your system (mosquitto.php in /var/www/html/protected and the others in  /usr/lib/cgi-bin/ecu/) you can refresh the mosquitto configuration page and check the retain flag. 
Important: the permissions on the files should be equal to the original files. 
Please let me now if it works as expected.

closed 2023-02-15 by patience4711