patience4711/RPI-APS-inverters #83: Error in power calculation?

peter33826, 2023-03-01:
Attached a screenshot of my device. It works great, Thanks a lot for this. But maybe there is an error in power calculation:
40,9V x 4,2A is not 124,9 W
32,8V x 2,3A is not 90,1 W

[image]

Thanks for any ideas.

patience4711, 2023-03-03:
The polling has a frquency of once in 5 minutes. The calculation of some values can be found in the wiki (subject polling). It is obvious that you can't use the u*i formula since these are momentary values that fluctuate a lot within 5 minutes.

peter33826, 2023-03-03:
I expected a 5 min (moving) average for U and I and then calculation of P. Thank you for the explanation.

closed 2023-03-03 by peter33826