patience4711/RPI-APS-inverters #88: Problems with HTTP Request

HackTheMac, 2023-07-02:
Wehen using the HTTP Request to retrieve date, e.g. http://rpiecu/recovery/getInverter.php?inv=1, the JSON is not directly returned. Rather there is some header, e.g. the header tag before the JSON which makes processing rather inconvenient.

patience4711, 2023-07-03:
Thank you for your interest in my software. What version are you running? Are you new to this software? Are you able to login on the rpi and edit files? What browser do you use?

I really can't understand what you get, but I get this (as intended) with Chrome and Edge: 

[image]

This is a copy of the file invData0 in ramdisk. If that file is corrupt, the webinterface wouldn't show values. 

If you can edit /var/www/html/recovery/getInverter.php you could add `</body></html>` at the end, maybe that helps.

HackTheMac, 2023-07-03:
Hi patientce4711,
no, I am not new to your software. I have been using it for some time and I am very impressed by what you are delivering.
I am capable of logging into the rpi and edit files. I am sorry I am not good with GitHub that is probably why you are not getting my issue. I fully agree that when calling getInverter.php a perfectly fine looking json string shows up in the browser. However, if you take a look of the source code of said displayed page you will note that there is more than just the json string. As previously mentioned, there is a header tag prior to the string.

I just looked at getInverter.php:
[image]
While I am not good with PHP, I would assume that the first three lines is what is causing my issue. 

This is not severe. I just wanted to give you some feedback.

patience4711, 2023-07-03:
Yes, if you ask for the page source you get  the page source. That is normal html. However, you can remove the html part so that it starts with `<?php` en ends with `?>` 
Then the page source is only the echoed json.

closed 2023-07-14 by HackTheMac