patience4711/RPI-APS-inverters #92: Strange things with only 1 panel..

defaliz, 2023-10-01:
Hi,
I use rpi ecu on pi zero for one year without problem..
Now I've add a new ds3 controller with one panel..
And there is a strange thing on the new  inverter datas.
The voltage is good on the number 2 panel, but there is no panel..!
And the voltage is bas on the first panel, 9,9v, which is plugged..
What's the problem ?
Regards
[image]

DrLabel, 2023-10-01:
I have the same problem 
[image]

defaliz, 2023-10-01:
Hi Dr label..interesting...
 It's better for you because your dc amp is 9A, mine is 0,1..
Have you try with 2 panels ?

DrLabel, 2023-10-01:
I have 2 260w panels in parallel connect to one input, I'm waiting for 1 450w panel to connect to the other input

defaliz, 2023-10-01:
[image]

I repost the picture beacause in the previous, everybody see my ip address..

patience4711, 2023-10-06:
This is a strange behaviour. Could it be that in the inverterconfiguration both the panels are checked? You should uncheck one of them.

defaliz, 2023-10-06:
Here is the result of a polling of inverter 2..
running script inverterPoll.pl...
argument is 2
sudo /usr/lib/cgi-bin/ecu/inverterPoll.cgi 2
Content-type:text/html
running inverterPoll.cgi
inverterpoll running with arg 2
invSerial from file = 703000654290
invType = 2
idx = 0
invIdrev = 70C7
idShort = xxxx
idinvers from file = 80971B01A3D8
poll cmd = 1D240170C71414060001000F1380971B01A3D8FBFB06BB000000000000C1FEFE9E
answer = FE0164010064FE034480001401D2FE0345C470C70035FE0345C470C70035FE7D44810000060170C71414006F00DB8086000069703000654290FBFB5CBBBB20000500CBFFFF0000000000008000074801DF000500000385138B595F00000001FFFF00000936000C7FA9000000000801FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF34A4FEFE70C700AF
messageToDecode = FE0164010064FE034480001401D2FE0345C470C70035FE0345C470C70035FE7D44810000060170C71414006F00DB8086000069703000654290FBFB5CBBBB20000500CBFFFF0000000000008000074801DF000500000385138B595F00000001FFFF00000936000C7FA9000000000801FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF34A4FEFE70C700AF
message split after 44810000 060170C71414006F00DB8086000069703000654290FBFB5CBBBB20000500CBFFFF0000000000008000074801DF000500000385138B595F00000001FFFF00000936000C7FA9000000000801FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF34A4FEFE70C700AF
polling an invertertype: 2
processing voltage & current
json from file = { 'heath': 22.8, 'dcv': [ 10.0, 38.9, 0.0, 0.0 ], 'freq': 50.0, 'totalen': 13.6, 'acv': 237.4, 'idx': 0, 'en': [ 13.6, 0.0, 0.0, 0.0 ], 'totalpw': 2.3, 'totalenStack': 13.6, 'dcc': [ 0.1, 0.0, 0.0, 0.0 ], 'enStack': [ 13.6, 0.0, 0.0, 0.0 ], 'time': 22863, 'pwr': [ 2.3, 0.0, 0.0, 0.0 ] }
time from file = 22863
en_old_total from file = 13.6
totalenStack from file = 13.6
time from inverter 22879
the timespan = 16
* * * * POLLED INVERTER 2 * * * *
** values panel 0**
extracted energy_ext[x] = 819113.000000
en_new[x] = 13.597275 en_old[x] = 13.600000
en_increase[x] = -0.002726
en_increase[x] invalid, ignoring this poll
all cmds sent
returned by sendCommands() 100
toLog =
15:00:35 polled inv 2 error 100 invalid energy value
logCmd = echo "
15:00:35 polled inv 2 error 100 invalid energy value" >> /ramdisk/ecu_log.txt
Content-type:text/html
running invdataNull.cgi
echo "{ 'heath': 0, 'dcv': [ 0, 0, 0, 0 ], 'freq': 0, 'totalen': 13.6, 'acv': 0, 'idx': 0, 'en': [ 13.6, 0.0, 0.0, 0.0 ], 'totalpw': 0, 'totalenStack': 13.6, 'dcc': [ 0, 0, 0, 0 ], 'enStack': [ 13.6, 0.0, 0.0, 0.0 ], 'time': 22863, 'pwr': [ 0, 0, 0, 0 ] }" > "/ramdisk/invData2"
sedCmd = sed -i 's|['\'']|'\"'|g' /ramdisk/invData2
invdataNull.cgi: resetted existing invData
invdataNull.cgi exit with code 0
inverterpoll exit with code 0
command processed
HTTP:1/1 200 OK

defaliz, 2023-10-06:
Hi, thank for reply
in the beginning i've only configure one panel, and the value was 
dc Volts 10 v dc Amp 0.1 (my aps inverter is new)
so i've try to configure a second panel (but not physically connected)..
and surprise ! the value was the same for the first panel, but for the second they where : 
dc Volts 38 V dc Amp 0
it's seems that the soft is mixing the values of the two panels...
can you reproduce this by unplugging and disabling a panel on your installation ... ? (i can't make more tests because my installation is 200km far away..)
regards
Alain

PS :Here is the result of a polling of inverter 2..
running script inverterPoll.pl...
argument is 2
sudo /usr/lib/cgi-bin/ecu/inverterPoll.cgi 2
Content-type:text/html
running inverterPoll.cgi
inverterpoll running with arg 2
invSerial from file = 703000654290
invType = 2
idx = 0
invIdrev = 70C7
idShort = xxxx
idinvers from file = 80971B01A3D8
poll cmd = 1D240170C71414060001000F1380971B01A3D8FBFB06BB000000000000C1FEFE9E
answer = FE0164010064FE034480001401D2FE0345C470C70035FE0345C470C70035FE7D44810000060170C71414006F00DB8086000069703000654290FBFB5CBBBB20000500CBFFFF0000000000008000074801DF000500000385138B595F00000001FFFF00000936000C7FA9000000000801FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF34A4FEFE70C700AF
messageToDecode = FE0164010064FE034480001401D2FE0345C470C70035FE0345C470C70035FE7D44810000060170C71414006F00DB8086000069703000654290FBFB5CBBBB20000500CBFFFF0000000000008000074801DF000500000385138B595F00000001FFFF00000936000C7FA9000000000801FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF34A4FEFE70C700AF
message split after 44810000 060170C71414006F00DB8086000069703000654290FBFB5CBBBB20000500CBFFFF0000000000008000074801DF000500000385138B595F00000001FFFF00000936000C7FA9000000000801FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF34A4FEFE70C700AF
polling an invertertype: 2
processing voltage & current
json from file = { 'heath': 22.8, 'dcv': [ 10.0, 38.9, 0.0, 0.0 ], 'freq': 50.0, 'totalen': 13.6, 'acv': 237.4, 'idx': 0, 'en': [ 13.6, 0.0, 0.0, 0.0 ], 'totalpw': 2.3, 'totalenStack': 13.6, 'dcc': [ 0.1, 0.0, 0.0, 0.0 ], 'enStack': [ 13.6, 0.0, 0.0, 0.0 ], 'time': 22863, 'pwr': [ 2.3, 0.0, 0.0, 0.0 ] }
time from file = 22863
en_old_total from file = 13.6
totalenStack from file = 13.6
time from inverter 22879
the timespan = 16
* * * * POLLED INVERTER 2 * * * *
** values panel 0**
extracted energy_ext[x] = 819113.000000
en_new[x] = 13.597275 en_old[x] = 13.600000
en_increase[x] = -0.002726
en_increase[x] invalid, ignoring this poll
all cmds sent
returned by sendCommands() 100
toLog =
15:00:35 polled inv 2 error 100 invalid energy value
logCmd = echo "
15:00:35 polled inv 2 error 100 invalid energy value" >> /ramdisk/ecu_log.txt
Content-type:text/html
running invdataNull.cgi
echo "{ 'heath': 0, 'dcv': [ 0, 0, 0, 0 ], 'freq': 0, 'totalen': 13.6, 'acv': 0, 'idx': 0, 'en': [ 13.6, 0.0, 0.0, 0.0 ], 'totalpw': 0, 'totalenStack': 13.6, 'dcc': [ 0, 0, 0, 0 ], 'enStack': [ 13.6, 0.0, 0.0, 0.0 ], 'time': 22863, 'pwr': [ 0, 0, 0, 0 ] }" > "/ramdisk/invData2"
sedCmd = sed -i 's|['\'']|'\"'|g' /ramdisk/invData2
invdataNull.cgi: resetted existing invData
invdataNull.cgi exit with code 0
inverterpoll exit with code 0
command processed
HTTP:1/1 200 OK

patience4711, 2023-10-09:
Hi, yes if found the bug, as far as i can see it the only problem is that the dcv values are  swapped for a ds3. This was simple to correct, i will issue a software update for this soon...

defaliz, 2023-10-09:
Great Mr Hansiart !
it's a very good new !
thank for the work..
regards

defaliz, 2023-10-25:
hi, 
thank you very much for correction...
(in the readme file date is still in 2022 "2022-10-09")
for updating and corrected the bug do i need to change all the files ? because i've made customization in /var/www/html
is it possible to change only ecu folder ?
best regards

defaliz, 2023-10-31:
hi,
i've update rpiecu.. it's better but...
the DC amp is very low... (0,1 "porte inv2") the others panels (milieu inv2 and mur inv1 ) 
are just beside and they have good value (but they have 2 panels)..
do you know why ?
regards
[image]
[image]

patience4711, 2023-10-31:
Sorry no, i don't think there is another bug like a swab of the panels in the software, otherwise we would se some value for panel 2. No one else has reported this problem. 
For a start you could change the inverterconfiguration in the software such that there is only one panel. I think the inverter should work with only one panel. Things you could try is measuring the output with a power meter or connect the panel to the other channel. Good luck. Maybe it just needs a reset and is it oke tomorrow.

defaliz, 2023-11-01:
thanks for your help...
it's better now... i've unchecked panel 2 on the screen you can see "porteinv2" with 11,2 w for one panel (no sun in Normandy and we are waiting for a big storm !!!)
how are calculate power ? because with inverter 1 i have 

panel | dc Volts | dc Amp | power W | nrg Wh
-- | -- | -- | -- | --
1 | 36.4 | 0.6 | 25.9 | 33.8

so power = 36,4 x 0,6 = 21,84  w and not  25.9 w ?
regards

patience4711, 2023-11-02:
Glad that it works. The extraction and calculation of values is pointed out [here](https://github.com/patience4711/RPI-APS-inverters/wiki/3-INVERTERS#polling).

defaliz, 2023-11-02:
many thanks !

closed 2023-11-06 by patience4711