patience4711/RPI-APS-inverters #93: Raspi not reachable but in my network with IP Adress

TheRealCryss, 2023-10-06:
Hello!

I try to get this working on a Raspberry Zero, but after the initial WLAN setup (Over WLAN of course) and booting afterwards nothing happens.

I have a monitor connected to it and the only issue from the boot-up messages is [FAILED] Failed to start dhcp on all interfaces.

The login prompt comes up for a few seconds and vanishes with only a blinking courses instead. The Raspi is connected to my network and got its IP according to my Fritzbox router, but is neither reachable from http nor ssh nor can I ping it (time out). The CPU gets quite hot and after 1 hours still nothing happens.

I can put the SD-card back to my Pi 3 and it works.

Do you have any ideas?

patience4711, 2023-10-09:
Sorry no clue. You say that the software works on a pi3 ? Is your ip of the family 192.168.x.x ? With a terminal connected you should be able to configure the network by hand.

TheRealCryss, 2023-10-10:
Hi! Thanks for the response.
I had two SD cards and both showed the same error. I bought another one and first had the problem again. But somehow it is working now and I can't tell for sure what it was.

TheRealCryss, 2023-10-11:
I close this issue as somehow it works now.

closed 2023-10-11 by TheRealCryss