patience4711/RPI-APS-inverters #94: I can't pair with the inverter

TheRealCryss, 2023-10-11:
After I changed the Raspberry computer from a Pi 3 to a Pi Zero W I can't pair the converter anymore and I don't know what the messages mean:

Pairing command
```
running script inverterPair.pl...
argument is 0
sudo /usr/lib/cgi-bin/ecu/inverterPair.cgi 0
Content-type:text/html
running inverterPair.cgi

inverterpair running with arg0
wiringPi started
opened port serial0
fileToread = /var/www/ecu_data/ecuProperties
json from file = { 'idinvers': "80971B01A3D8", 'id': "D8A3011B9780" }
idinvers from file = 80971B01A3D8
ecushort = A3D8
fileToread = /var/www/ecu_data/inverters/invProperties0
json from file = { 'panels': [ 1, 1, 0, 0 ], 'idx': "895", 'id': "unpaired", 'name': "Inv1", 'type': 2, 'serial': 702000014595 } serial from file = 702000014595
trying to pair inverter0
send cmd 0 2524020FFFFFFFFFFFFFFFFF14FFFF140D0200000F1100702000014595FFFF10FFFF80971B01A3D8FA
info = 4
no incoming messages
pairing failed, try again...
inverterPair ready
pair command processed
HTTP:1/1 200 OK
```

Start coordinator for pairing
```
running script startCoordinator.pl...
calling startCoordinator.cgi argument is 0
command to run = sudo /usr/lib/cgi-bin/ecu/checkCoordinator.cgi
calling checkCoordinator.cgi Content-type:text/html
running checkCoordinator.cgi

wiringPi started
opened port serial0
command = 00270027
no answer
no answer on serial
toLog =
16:09:56 checking zb system failed, no answer on serial
logCmd = echo "
16:09:56 checking zb system failed, no answer on serial" >> /ramdisk/ecu_log.txt
result = 256
coordinator checked, not running, starting
command to run = sudo /usr/lib/cgi-bin/ecu/startCoordinator.cgi 0
calling startCoordinator.cgi Content-type:text/html
running startCoordinator.cgi

startCoordinator running with arg 0
wiringPi started
opened port serial0
fileToread = /var/www/ecu_data/ecuProperties
idinvers from file = 80971B01A3D8
ecushort = D8A3
base commands edited
cmd = 03260503010321
answer = FE064180010202020702C1
cmd = 0141000040
no answer
cmd = 0A26050108FFFF80971B01A3D856
no answer
cmd = 032605870100A6
no answer
cmd = 0426058302D8A3DD
no answer
cmd = 062605840400000100A4
no answer
cmd = 0D240014050F0001010002000015000020
no answer
cmd = 00260026
no answer
all cmds done
coordinator should be running now
command to run = sudo /usr/lib/cgi-bin/ecu/checkCoordinator.cgi
calling checkCoordinator.cgi Content-type:text/html
running checkCoordinator.cgi

wiringPi started
opened port serial0
command = 00270027
no answer
no answer on serial
toLog =
16:10:21 checking zb system failed, no answer on serial
logCmd = echo "
16:10:21 checking zb system failed, no answer on serial" >> /ramdisk/ecu_log.txt
result = 256
coordinator checked, not running, starting
command to run = sudo /usr/lib/cgi-bin/ecu/startCoordinator.cgi 0
calling startCoordinator.cgi Content-type:text/html
running startCoordinator.cgi

startCoordinator running with arg 0
wiringPi started
opened port serial0
fileToread = /var/www/ecu_data/ecuProperties
idinvers from file = 80971B01A3D8
ecushort = D8A3
base commands edited
cmd = 03260503010321
answer = FE064180010202020702C1
cmd = 0141000040
no answer
cmd = 0A26050108FFFF80971B01A3D856
no answer
cmd = 032605870100A6
no answer
cmd = 0426058302D8A3DD
no answer
cmd = 062605840400000100A4
no answer
cmd = 0D240014050F0001010002000015000020
no answer
cmd = 00260026
no answer
all cmds done
coordinator should be running now
command to run = sudo /usr/lib/cgi-bin/ecu/checkCoordinator.cgi
calling checkCoordinator.cgi Content-type:text/html
running checkCoordinator.cgi

wiringPi started
opened port serial0
command = 00270027
no answer
no answer on serial
toLog =
16:10:45 checking zb system failed, no answer on serial
logCmd = echo "
16:10:45 checking zb system failed, no answer on serial" >> /ramdisk/ecu_log.txt
result = 256
coordinator checked, not running, starting
command to run = sudo /usr/lib/cgi-bin/ecu/startCoordinator.cgi 0
calling startCoordinator.cgi Content-type:text/html
running startCoordinator.cgi

startCoordinator running with arg 0
wiringPi started
opened port serial0
fileToread = /var/www/ecu_data/ecuProperties
idinvers from file = 80971B01A3D8
ecushort = D8A3
base commands edited
cmd = 03260503010321
answer = FE064180010202020702C1
cmd = 0141000040
no answer
cmd = 0A26050108FFFF80971B01A3D856
no answer
cmd = 032605870100A6
no answer
cmd = 0426058302D8A3DD
no answer
cmd = 062605840400000100A4
no answer
cmd = 0D240014050F0001010002000015000020
no answer
cmd = 00260026
no answer
all cmds done
coordinator should be running now
make /ramdisk/invChoice and set value and permissions

setPermissions.sh : checked permissions OK
script setPermissions.sh ready
HTTP:1/1 200 OK

startCoordinator runned
HTTP:1/1 200 OK
```

Maybe someone can say whether the zigbee module is bad or at what point it fails.
The inverter works, I have tested it with the official ECU-B router.

patience4711, 2023-10-11:
no answer means no answer, Check the wiring of the zigbee, maybe read the wiki to find out what tests you can do to debug

TheRealCryss, 2023-10-13:
I got the wiring wrong. I did it according to:
[image]

But GRD is wrong.

Then I looked up the Raspi GPIO-Pin layout from another source. I did the wiring according to:
[image]

And it all just works. I must have found out a year ago when I first set it up, but totally forgot about it.

patience4711, 2023-10-14:
That is the wiring for flashing the zigbee, not for the normal operation. This is also pointed out in the wiki [here](https://github.com/patience4711/RPI-APS-inverters/wiki/2-ZIGBEE-MODULE#operation). 
Anyway, glad to hear that it works now.

closed 2023-10-14 by patience4711