patience4711/RPI-APS-inverters #96: RPI-ECU : No values returned from QS1

autourdupc, 2023-11-29:
Hello.

Just created the module RPI-ECU with a PI zero 2 W wired to a CC2530, and paired my inverter QS1.
No data are returned.

Pooling the inverter give me thoses details.

running script inverterPoll.pl...
argument is 0
sudo /usr/lib/cgi-bin/ecu/inverterPoll.cgi 0
Content-type:text/html
running inverterPoll.cgi
inverterpoll running with arg 0
invSerial from file = 801000055441
invType = 1
idx = 5
invIdrev = 146C
idShort = xxxx
idinvers from file = 80971B01B3D6
poll cmd = 1D2401146C1414060001000F1380971B01B3D6FBFB06BB000000000000C1FEFE4F
answer = FE0164010064FE034480CD14011F
messageToDecode = FE0164010064FE034480CD14011F
no AF_DATA_CONFIRM
all cmds sent
returned by sendCommands() 11
toLog =
09:32:15 polled inv 0 error 11 AF_DATA_CONFIRM fail
logCmd = echo "
09:32:15 polled inv 0 error 11 AF_DATA_CONFIRM fail" >> /ramdisk/ecu_log.txt
Content-type:text/html
running invdataNull.cgi
echo "{ 'dcc': [ 0, 0, 0, 0 ], 'totalpw': 0, 'freq': 0, 'totalen': 0, 'acv': 0, 'heath': 0, 'dcv': [ 0, 0, 0, 0 ], 'idx': 0, 'en': [ 0, 0, 0, 0 ], 'pwr': [ 0, 0, 0, 0 ], 'time': 0 }" > "/ramdisk/invData0"
sedCmd = sed -i 's|[''']|'"'|g' /ramdisk/invData0
invdataNull.cgi: resetted existing invData
invdataNull.cgi exit with code 0
inverterpoll exit with code 0
command processed
HTTP:1/1 200 OK

Any idea ?

I alos would like to know the login / pass to access the PI via SSH.

autourdupc, 2023-11-29:
Will answer myself.

After too many attempts, I cleared database and try to add / pair again the inverter -> unsuccessfull
After a reboot, tryed to add / pair one more time the inverter and it worked !
Can now get data from inverted.

autourdupc, 2023-12-03:
Got again the same issue.

I tried to clear database, pair again -> Fail 
I spent too much time trying to make it running -> Fail each time.

Any idea ?

autourdupc, 2023-12-06:
Today, it stopped polling again...
[image]

kadzsol, 2023-12-08:
> @kadzsol Thanks for replying. Already tested, not working !
> 
> What I got again is polling failed today even without APS ECU running... Got only 3 points then polling failed. This is very strange.

More tips:

1. Make sure the RPI-ECU uses the same ECU ID as your APS-ECU
2. Remove connection TX pin of your RPI-ECU so it only listens to the inverter replies triggered by your APS-ECU
3. Set up a sniffer to see what is exactly happening: https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#_1-flashing-the-cc2531-stick

autourdupc, 2023-12-08:
> 3\. Set up a sniffer to see what is exactly happening:

Hum... Will need more time... Need to buy cc2531 stick

autourdupc, 2023-12-08:
Note also that even without APS-ECU powered off m'y RPI-ECU Lost something and polling comes fail.

patience4712, 2023-12-10:
The rpi-ecu sends a request and then waits some time for the answer. If nothing received within that time, too bad, will try again after next 5 minutes. Whether this conflicts with the aps-ecu i would't know, it shouldn't when they have the same id. I don't know however  if they both create a separate zigbee network because I assume you can have only one coordinator per network. So if you have just one network with a certain ID and there are 2 devices acting as coordinator.... maybe that doesn't work. But I'm just guessing because I don't have any deeper knowledge of Zigbee.  

When the polling isn't stable it is most likely a range problem, enough to be found on this in the issues. I don't know the weather in Paris these days, probably bad so maybe the inverter stops working when there is not enough light.

autourdupc, 2023-12-10:
@patience4712 
Thank you so much for replying.

I have stopped the APS ECU and only work with RPI-ECU.
I'm actually not able to make it polling again none of my inverters... QS1 or YC600.
Both can pair, but none of them is polling.
Very very strange.
When I try to pair again, YC600 always says "Paired successfull", while QS1 says always "fail".

I'm really very disapointed because I don't know where to search for issue... 
I can ssh the RPI-ECU but really don't know what part of the progra can help me debbugging. Any idea ?

kadzsol, 2023-12-10:
> The rpi-ecu sends a request and then waits some time for the answer. If nothing received within that time, too bad, will try again after next 5 minutes.

Maybe implement an option to wait indefinetly and let the real ECU initiate polling? Sort of stealth mode for people wants to use both devices at the same time.

> Whether this conflicts with the aps-ecu i would't know, it shouldn't when they have the same id. I don't know however if they both create a separate zigbee network because I assume you can have only [...]

APS does not implement a standard zigbee application, so having 2 coordinators in the same network is not a problem.

When you pair, the ECU defines the network ID for the inverter(s) to be paired. The network ID is not random but equals to the first 2 bytes of the ECU Id. Using the same ECU Id for both ECU's will mean that all the paired inverters (and ECU's) will be in the same network and can communicate.

When both ECU's are sending queries in the same network an issue may occur as the internal application sequence counters will mismatch. Let me explain what I mean with this. 

When a coordinator is started, it sends the first message with counter 1. Next message will get counter 2, etc. If the coordinators (ECU's) are not started at the same time, their application counters will be different. Also, a coordinator can send more or less messages, depending on many factors, making it sure that the application counters will misalign, even when you start them exactly at the same time. Devices (inverters) getting messages with older counters will disregard them. Hence my suggestion to let the real ECU do the query and let the RPI ECU only listen to whatever comes in.
 
> When the polling isn't stable it is most likely a range problem, enough to be found on this in the issues. I don't know the weather in Paris these days, probably bad so maybe the inverter stops working when there is not enough light.

True. Range and lack of sunlight can also impact the test.

autourdupc, 2023-12-11:
> I don't realy see the point of using 2 ecu's

@patience4712 
Hello.
This was only for the time to check that everything is working well before stopping APS-ECU...
Sure, no need to get them both running at the same time.
But in my case, while RPI-ECU is not working, lucky to get data via APS-ECU.

kadzsol, 2023-12-11:
> Hi kadszol, thanks for your clear explanation. I guess it's possible to have both ecu's working but I have no plans to change the polling procedure for this exceptional situation. I don't realy see the point of using 2 ecu's and I couldn't test anyway. Apart from that, I don't have time for it, too busy with other projects....

I can forsee situations when this function wouldbe useful. Imagine, you have a support issue with APS and want to upload the data to their cloud for further investigation. 

At the same time, you want to keep using your own setup to feed the same data to your home automation system.

patience4711, 2023-12-11:
But RPI-ECU does work. there are enough satisfied users to prove this. With the RPI zero W at least. 
Again, it is likely a range problem, move the ecu closer to the inverter. Bad wiring, maybe the zb antenna too close to the wifi of the rpi, insufficient daylight, the are multiple possible causes. 
When it works for others it should work for you too.

autourdupc, 2023-12-13:
> In my experience this all works better on a bright sunny day, Sometimes it seems as if pairing is made more difficult by bad weather conditions. There shouldn't be a reason for this but still....

Hello.
Today is a partial sunny day, and pairing worked !
As you said, need sun to pair.

Concerning RPI-ECU id, one said use same as APS-ECU, and you said do not use the same... What should I do ?

kadzsol, 2023-12-13:
@autourdupc 

If you do not use the same ID, they will not be able to communicate with the inverters at the same time. 

If you do use the same ID but let both ECU's query the inverters, the application might get confused. 

Unless one ECU is quering the inverters and the other ECU is only listening. Hope it is clear.

autourdupc, 2023-12-13:
> @autourdupc
> 
>Hope it is clear.
Sure !
Many thanks for the time spent for me.
Thanks to all of you.

patience4711, 2023-12-14:
Glad that it works now but it would be helpfull to know what exactly you changed. Did you move the ecu closer to the inverters?, what ecu-id did you use? ,  anything else? Or was it just the bad weatherconditions.

autourdupc, 2023-12-16:
> Glad that it works now but it would be helpfull to know what exactly you changed. Did you move the ecu closer to the inverters?, what ecu-id did you use? , anything else? Or was it just the bad weatherconditions.

It was just weather conditions...
When sun is shining, means panels give a lot of power, polling comes good.
This is the main difference with original APS-ECU that can poll even without any power from panels.

kadzsol, 2023-12-16:
@autourdupc 

Depends on your radio device. A cheap cc2531 without external antenne has very poor range. 

A CC2530 with an amplifier chip (CC259x) has excellent range and costs only a couple of more euro's. 

I have a CC2530+CC2591 and does not have a range issue.

autourdupc, 2023-12-16:
@kadzsol 
Concerning range, I'm at less than 5 meters for tests.
I will buy a better cc2530 with amplifier when I got everything running.

I have downloaded V3.9 and flashed my SD card.
Starting from zero.
Today is cloudy, and got only poor 21 watts from my QS1 with 4 panels ! It was enough to pair and get data from the inverter.
YC600 got only 1 panel and I could pair but unable to poll !

Everything seems to work correctly except YC600. Need to wait for better conditions.
I also then need to test the panels order to verify if they match their position. I let you know

patience4711, 2023-12-22:
I am closing this because its mixed-up with other issues

closed 2023-12-22 by patience4711