patience4711/RPI-APS-inverters #100: YC600 : inputs are swapped

autourdupc, 2023-12-13:
Hello.

I found that I connected my solar pannel to the input #1, but RPI-ECU give power for input #2.
See attached a photo of the linked input, and the screen shot of the measures.
[image]
[image]

autourdupc, 2023-12-13:
Will check... If got time tomorrow and got sun !

patience4711, 2023-12-14:
Ok great, thanks. The best way to test is unplug all panels and in the inverter settings only check one panel, starting with panel 1. Now plug a panel in channel 1 and see for which panel you get data. (after at least 2 polls).  Via the console you can poll manually. And repeat this for the other chanels.

Please make sure that you haven't checked a non-existing panel in the settings. 

It is btw a complete mystery to me that we are having this discussion. This chanel order was well researched at the time and has existed for years, also in the ESP-ECU project that has a lot of users worldwide. No one has ever noticed this, so it seems not very important. For the YC600 , if you put a sticker over channel 2 making it channel 0, the problem would be solved. Why didn't APS start from left to right with their numbering anyway? Adapting the software however, is a lot more complex. But if its wrong we have to fix it of course.
I am curious about the test results.

autourdupc, 2023-12-14:
> Would it be possible for you to replace /usr/lib/cgi-bin/ecu/inverterPoll.cgi with this new file?

I just uploaded the new file and give permissions.
Need to wait tomorrow some sun for the tests.
I will now change the settings in RPI-ECU (check panel #1 only) and let the polling process do the job tomorrow.

I let you know.

autourdupc, 2023-12-17:
@patience4711 
Hello.
Using V3_9 fresh install... RPI-ECU works fine !
Still got issues polling inverters, but comes from low power due to weather conditions.
As soon as the sun comes back, polling comes good (this is strange because APS-ECU still working whatever weather conditions are, or maybe they lie on values displayed).
I also noticed that when polling is not good (we can see "fail" while manual polling), the file "InvData" corresponding to the inverter is not created - Maybe this is by design.

I also got all the buttons, and inverters names match the correct inverters when using MS EDGE.
When using Firefox, still got some troubles, but it seems to come from navigator and not RPI-ECU.

Concerning panels orders, APSystem strategy is very strange...
YC600 
Panel 1 match input 1
Panel 2 match input 2
 -> Panel order is good

QS1
Panel 1 match input 2
Panel 2 match input 1
Panel 3 match input 3
Panel 4 match input 4
-> Only #1 and #2 are swapped - #3 and #4 are good.

Hope I could contribute a little to this great project.

kadzsol, 2023-12-17:
Regarding "working hours" difference of RPI/APS ECU's:

When an inverter has just enough sunlight to operate but not able to produce power, it will only send zigbee links state messages.  The APS ECU uses these messages to display inverter is on. I would expect it shows you 0W production. 

[image]

In the screenshot above, these are the time periodes between 9:00-10:15 and 14:35-15:40. 

If the RPI ECU would use those link state messages to indicate  inverter is on but power production is 0, you would probably see the same response in both ECU's unless one of the ECU's has a range problem.

FYI Zigbee link state message for inverter C28C/408000xxxxxx on network 199A. 

[image]

autourdupc, 2023-12-17:
@kadzsol 
This is very interesting...
However, I'm not enought aware of Zigbee commands... In the past I worked on MOD-BUS/J-BUS driver, and I got so much time :)
Today, I do not have enougth time to learn Zigbee. 
Sure, if RPI-ECU could use the same messages we would get the same graph, and could poll more often.
Thanks for the technical details. Hope one day I will be able to work on it (I just received my CC debugger CC2531 Zigbee USB dongle).

patience4711, 2023-12-18:
The aps inverters are known for their very low startup threshold, even on this very grey morning no poll failed starting 5 min. before sunrise (yc600). Local circumstances will influence this of course but to me it seems not very important.

Anyway, these zb signals are the mean reason why the system is not constantly reading incoming trafic. This program originally ran on an esp8266 with limited resources. I didn't know what would happen when say 10 inverters spitout beacons, link state signals, data, etc. And if there are other zigbee networks nearby, those signals are maybe received too.  So in theory the esp would get very busy evaluating all that traffic. The solution was to only listen for some time after a request was made. 

@autourdupc  Sure you contributed, thanks. So after all only the panels of the yc600 were swapped. I'am going to fix that now.

kadzsol, 2023-12-18:
> Anyway, these zb signals are the mean reason why the system is not constantly reading incoming trafic. This program originally ran on an esp8266 with limited resources. I didn't know what would happen when say 10 inverters spitout beacons, link state signals, data, etc. And if there are other zigbee networks nearby, those signals are maybe received too. So in theory the esp would get very busy evaluating all that traffic. The solution seemed to only listen for some time after a request was made.

The application only receives messages that are ment for their own zigbee network. 

If messages are transmitted for other networks, the MAC/Network layer of the protocol will take care of the filtering. 

APS also uses ESP's in their ECU's so nothing wrong with those devices.

patience4711, 2023-12-18:
No, nothing wrong with ESP's.  But I don't know if you can compare the Esp in an APS ECU with a nodemcu. Could be that there is larger memory chip or an extra external memory. But even when they are the same, i suspect that APS-ECU uses the esp mainly to upload data to their cloud server where it is processed and served. That would make a great difference in cpu and program space requirements.

kadzsol, 2023-12-19:
Some people opened their ECU's and inverters and posted the photo's on internet. They seem to use the same 2 euro/dollar ESP's we use too :-) 

You are correct regarding calculation. The inverters and the ECU's just passing on data to their cloud and calculation happens there or in your app.

autourdupc, 2023-12-22:
@kadzsol 
Hello.

I would like to know wich firmware should I use to flash CC2530+CC2591...
Some say that it is a special firmware that can manage CC2591.

I tried to flash with bundled "CC2530ZNP-with-SBL.hex" file but module CC2530+CC2591 seem not work while the CC2530 module do work perfectly.

kadzsol, 2023-12-22:
> @kadzsol Hello.
> 
> I would like to know wich firmware should I use to flash CC2530+CC2591... Some say that it is a special firmware that can manage CC2591.
> 
> I tried to flash with bundled "CC2530ZNP-with-SBL.hex" file but module CC2530+CC2591 seem not work while the CC2530 module do work perfectly.

[cc2530_test.zip](https://github.com/patience4711/RPI-APS-inverters/files/13751713/cc2530_test.zip)
use CC2530ZNP_2591-with-SBL.hex from the zip above

autourdupc, 2023-12-22:
@kadzsol 
I think I'll never be lucky on this project.
Only my CC2530 is working... My CC2530+CC2591 don't want to work... I use my new CC debugger to flash it.

kadzsol, 2023-12-22:
I use the same device so it can be done.

1. did flashing with correct file completed without errors?
2. use the same procedure/device to flash as used for your working radio device
3. replace only the working radio device with the new one (so no other rpi, wires, psu)
4. hook up zigbee sniffer

autourdupc, 2023-12-22:
Sure... The issue comes from my side.
What is strange is that now I got data, while the test with 2710AABBCC still fail... very strange !
I tested power value on module... Got 3,34 Vdc so its good... No loss
I will continue to investigate

kadzsol, 2023-12-22:
I do not know what 2710AABBCC is... but good that it works.

patience4711, 2023-12-22:
it is the loopback test (with preceding FE and length), should work with a cc2530
when i run it on my system i get the data AABBCC back:
_running script sendZigbee.pl...
argument is 2710AABBCC
sudo /usr/lib/cgi-bin/ecu/testZigbee.cgi 2710AABBCC
Content-type:text/html
running testZigbee.cgi
testZigbee running with arg 2710AABBCC
wiringPi started
slen = 03
command incl. FE, len and crc = FE032710AABBCCE9
inMessage = FE036710AABBCCA9
command processed
HTTP:1/1 200 OK_

autourdupc, 2023-12-22:
My Loopback test... Very strange... I can collect data but loopback is wrong.

_running script sendZigbee.pl...
argument is 2710AABBCC
sudo /usr/lib/cgi-bin/ecu/testZigbee.cgi 2710AABBCC
Content-type:text/html
running testZigbee.cgi
testZigbee running with arg 2710AABBCC
wiringPi started
slen = 03
command incl. FE, len and crc = FE032710AABBCCD0
no answer
command processed
HTTP:1/1 200 OK__

patience4711, 2023-12-22:
I see that you have a different checksum which seems impossible to me. No matter what stupid things i do (adding spaces fi) i always get checksum E9. 
Try 2710AA should give  
testZigbee running with arg 2710AA
wiringPi started
slen = 01
command incl. FE, len and crc = FE012710AA9C
inMessage = FE016710AADC
command processed
HTTP:1/1 200 OK

kadzsol, 2023-12-22:
Tested with my other (non-PR) CC2530+2591 and it responds just fine:

66: Sent=FE032710AABBCCE9
67: Received=FE036710AABBCCA9

autourdupc, 2023-12-22:
That is crazy...

testZigbee running with arg 2710AA
wiringPi started
slen = 01
command incl. FE, len and crc = FE012710AA87
no answer
command processed
HTTP:1/1 200 OK

Maybe the firmware of the CC2530+2591 ?

patience4711, 2023-12-22:
No, it's not the firmware, see the post of @kadzsol. 
We are using the same software on our raspberry no? So if i input 2710AA via the console, the command after adding FE,  length and crc is  FE012710AA9C but you are getting FE012710AA87.  So the same script calculates a different checksum. That is not possible! What are you doing??? Are you using the console to issue the command?

autourdupc, 2023-12-22:
I think I got black cats at home ! Or my home is on an old cemetery!!!
That is incredible...
I use the console to send commands.
I only modified PHP files for charts... Nothing to do with CGI.
Tomorrow I will again flash last version and redo the tests and let you know.

patience4711, 2023-12-23:
I can't believe that would help. There can't be a difference in the scripts itself so the way the input is processed is the same. The problem must be the interface. Did you try this with different browsers? And what happens if you provide wrong input, like 2710ABC which gives me:
**slen = 01
command incl. FE, len and crc = FE012710ABC9D
no answer**
Length is wrong because a byte is 2 hex characters.

Might be better to catch those ghosts and cats....

autourdupc, 2023-12-23:
@patience4711 
Why is the timestamp of the rpiecu-v3_9.img in the download archive dated 29/07/2022
While the previous rpiecu-v3_9.img was dated 15/12/2023 ?

Actual download
[image]

Previous download
[image]

autourdupc, 2023-12-23:
> And what happens if you provide wrong input, like 2710ABC which gives me:

unning script sendZigbee.pl...
argument is 2710ABC
sudo /usr/lib/cgi-bin/ecu/testZigbee.cgi 2710ABC
Content-type:text/html
running testZigbee.cgi
testZigbee running with arg 2710ABC
wiringPi started
slen = 01
command incl. FE, len and crc = FE012710ABC86
no answer
command processed
HTTP:1/1 200 OK

patience4711, 2023-12-23:
An image of a 16Gb has to be shrinked. This is done in a linux system, where the timestamp gets wrong probably by overwriting a filename or something. At that time the 3_7 image didn't exist. Don't worry, that is not the cause of your problem. 
I am making a new image, this time from the sd in my original system. I hope that the timestamp will be oke now. Just wait an hour or so.

autourdupc, 2023-12-23:
Hello.

Just FYI, after creating a new SDcard, with 3.9a and doing factory reset, got same issue with loopback...
I really think it comes from CC2530 
I will revert back hardware and use my CC2530 single ship.

Sorry bother you, but 3.9a do not connect ssh using default root password.

I also wish you happy holidays.

kadzsol, 2023-12-23:
> Just FYI, after creating a new SDcard, with 3.9a and doing factory reset, got same issue with loopback... I really think it comes from CC2530 I will revert back hardware and use my CC2530 single ship.

We do not think that, because the input string (crc =cyclic redundancy check, the last 2 characters) to your radio device is incorrect, so the firmware rejects it. This is the expected behaviour. 

With other words, the root cause we would expect to be somewhere "before" the radio device, in components/code that are responsible for creating the input command including the crc.

To rule out the radio device from the equitation, you may install Z-tool on a Windows PC (part of https://www.ti.com/tool/Z-STACK installation) and hook up your radio device via a USB-TTL adapter to test the loopback command.

autourdupc, 2023-12-23:
@kadzsol
It comes difficult... Z-Tool do not operate on my Win 10 64 computer... Remains hang with pop-up "Please wait".

Got also 2 modules CC2531, but pins are so small (1.25 mm or less), will continue with CC2530 modules.

Back to RPI-ECU
I tested commandline and got same issue, whatever I use CC2530 (was working on loopback) or CC2530+2591
-> We can consider interface is not faulty
And because got now 2 modules with same issue, we can say it come from Pi module.

I already flash another RPI-ECU version (other SD card) and got same issue. I come crazy.
I do not have another Pi Zero to swap hardware.

```
sudo /usr/lib/cgi-bin/ecu/testZigbee.cgi 2710AABBCC
Content-type:text/html

<html>
<head>
<title>RPI ECU PROGAM</title>
</head>
<body>
<h4>running testZigbee.cgi</h4>
testZigbee running with arg 2710AABBCC
<br>
wiringPi started
<br>
slen = 03
<br>
command incl. FE, len and crc = FE032710AABBCCD0
<br>
no answer
<br>
```

Can get values from panels, can issue poll commands, but loopback not working.

kadzsol, 2023-12-23:
> @kadzsol It comes difficult... Z-Tool do not operate on my Win 10 64 computer... Remains hang with pop-up "Please wait".

I have the same and have a black cat :-) Run it as administrator.

autourdupc, 2023-12-23:
> Run it as administrator.
Not better !

autourdupc, 2023-12-23:
@patience4711 
@kadzsol 

I reinstalled you RPI-ECU 3_3... It works with both CC2530 and CC2530+2591

testZigbee running with arg 2710AABBCC
wiringPi started
command incl. FE, len and crc = FE032710AABBCCE9
inMessage = FE036710AABBCCA9
command processed
HTTP:1/1 200 OK

It comes from the V3_9 / V3_9a !

patience4712, 2023-12-23:
The ssh pswd = pietje321
Consider this: the script testZigbee.cgi is exactly the same as on 3_3, i never touched it the last 2 years, it always worked and in my system (of which the image 3_9 comes) it works.  Never seen a crc problem, it is exactly calculated the way it is for a poll command and that works in your system.
So i am afraid that i cannot solve this but hey, if it works for the other part why bother.

autourdupc, 2023-12-24:
@patience4711 
Can't stay with something I can't understand... 

After many tests V3_3 vs V3_9a, and comparison of files after installing fresh SDcard image without modifying anything.
testZigbee.cgi is not the same on both versions...
File size and date are not the same.
[image]

 And finally, I replaced the file from V3_3 in the V3_9a folder and it works perfectly !

[image]

```
running script sendZigbee.pl...
argument is 2710AABBCC
sudo /usr/lib/cgi-bin/ecu/testZigbee.cgi 2710AABBCC
Content-type:text/html
running testZigbee.cgi
testZigbee running with arg 2710AABBCC
wiringPi started
command incl. FE, len and crc = FE032710AABBCCE9
inMessage = FE036710AABBCCA9
command processed
HTTP:1/1 200 OK
```

Your name is true... "Patience"... Sure, you are very patient.
Thanks you so much for all your great job and time you spent with me.

I wish you a Merry Christmas.

patience4712, 2023-12-24:
Blow the horn!, it crossed my mind that you should try that. It doesn't explain however why it works on my system (tested it yesterday again). I can't believe it myself now (probably you think I have ghosts too) but i am not at home to test ( for the 250th time).  
 After Chrismas I'l kook again what could have been wrong, I mean if you change a cgi you have to compile it so these things can't happen accidently. 
Merry Christmas to you too....

autourdupc, 2023-12-25:
@patience4711 
I try to understand how chart_pe.php is working...
I'm not very familiar with javascript nor json, but want to understand and play with it.
I saw some samples on chart-js.org for graphic charts, but i really do not understand where the data are collected from, and how they are collected.
I do not see the query that collect data from the selected day.
Any explanation possible ?

patience4712, 2023-12-26:
@autourdupc  I started to doubt myself, but the zigbee commands I had sent via the console got an answer so they have a correct crc. The only system i have with a zigbee module is my working v3_9a version so...... 

The query (from the influx database) is done by another php file that is included or called  so you should be able to trace that. I am still not at home so i can't tell much more right now.
Happy playing....

autourdupc, 2023-12-26:
> the zigbee commands I had sent via the console got an answer

@patience4712
I did the same, and was not working with 3_9a...Only 3_3 works ! So strange.

patience4711, 2023-12-27:
I just tested the zigbeecommand again on my 3_9 system, it works perfectly. 
To investigate i checked the sourcecode of testZigbee.cgi. The last modificationdate is april 22 2023. So i touched it indeed, now i remember a small efficiency improvement in the crc !! subroutine. The image v3_3 has the date 27 june so at the time this modification was already implemented.  That is why i think that the files of v3_3 and 3_9 cannot be different. 

Moreover, if you upgrade from 3_3 to 3_9, all cgi files of 3_3 are overwritten with the 3_9 files and that didn't work for you. 
The very same crc routine is used in all zigbeecommands, they all work but not when you input them yourself (via the console). 

I wrote the 3_3 image on an sd and run it on an other rpi zero. Tried the loopback command, it works as expected with the right crc.  When i compare the filezises there is a difference like you said. But they both work! 

Oke my conclusion is that it cannot be platform related otherwise the crc problem would arise in other zigbee commands as well. So it must be the input.
I ran in the ssh commandline 
cd /usr/lib/cgi-bin/ecu
./testZigbee 2710AABBCC and that works oke.
If you could do the same with [this file](

autourdupc, 2023-12-27:
Hello

Got in the same directory both testZigbee.cgi files... (thoses in previous screen copies above).
1) testZigbee.cgi -> Coming form SDcard V3_3
2) testZigbee.cgi_Original -> Coming from SDcard V3.9a 

[image]

Still not working with 3_9a
[image]

Black cat is elsewhere !

patience4711, 2023-12-27:
nonono, i asked you to run the **test_testZigbee.cgi** file that you can find [here](https://github.com/patience4711/RPI-APS-inverters/blob/main/test_testZigbee.cgi). So that we are absolutely sure we are running the same script with the same input.

autourdupc, 2023-12-27:
Sorry... Not better !

[image]

patience4711, 2023-12-28:
So we really have the phenomena here that exactly the same script with exactly the same input gives different results? Mine gives consequently E9, yours D0. That's impossible, incredible, crazy. I think we have to reconsider our knowledge about computerlogics completely.  

There is only one logical explanation possible; your "2710AABBCC" is not the same as my "2710AABBCC" but how that could differ.... i can't make it up.

I don't believe in ghosts so i must be overlooking something. To get more insight i recompiled the file with some more debug output, running. /zigbeeTest.cgi 2700AABBCC should give the result below.
[zigbeeTest.cgi](https://github.com/patience4711/RPI-APS-inverters/blob/main/zigbeeTest.cgi)
more tests: 
2710112233 should give crc 34 ( to check if we have the same error with only numbers).
2700 should give crc 27 (does this command work?)
[image]

autourdupc, 2023-12-28:
Hello.
I don't beleive in ghosts too...
[image]

patience4711, 2023-12-28:
Wel the error is in the crc subroutine. This is a piece of c code used in all situations where a zigbeecommand is sent. It is the same in the startcoordinator, and inverterpoll commands. And the same in all the ECU projects that run on an esp8266 or esp32. It always works, even for you (polling and coordinatorstart works).  But not in this specific case on your platform.
Input length and appearance is the same. Then it must be the platform.

There is one thing you could do to check if the platform is the pitfall; compile the file on your platform. The sourcecode is [testZigbee_new.cpp](https://github.com/patience4711/RPI-APS-inverters/blob/main/testZigbee_new.cpp).
If you copy that in /usr/lib/cgi-bin/ecu and run:
g++ -lcgicc testZigbee_new.cpp -lwiringPi -o testZigbee.cgi, Your testZigbee.cgi file will be overwritten. 

if you run it, it  shows<h4>running testZigbee.cgi version 2023-12-28</h4> ... etc

autourdupc, 2023-12-28:
I didn't wanted to overwite the existing file, so i changed the output name (same a source)...

[image]

patience4711, 2023-12-28:
oke, don't mind that warning.  So now ./testZigbee_new.cgi 2710AABBCC

autourdupc, 2023-12-28:
OK.
So please give me the compiled version and I will try to confirm it works

patience4711, 2023-12-28:
it's [here](https://github.com/patience4711/RPI-APS-inverters/blob/main/testZigbee_old.cgi)

autourdupc, 2023-12-28:
Arf !

[image]

patience4711, 2023-12-28:
and stil the original testZigbee.cgi from version 3_3 works?

autourdupc, 2023-12-28:
Sure !

[image]

patience4711, 2023-12-28:
One thing to check maybe: on the commandline type **mc** you'l get the midnight commander that you can use to cmod testZigbe_old.cgi, that should look like below: If not you can change it accordingly. And run it again.
[image]

One last experiment before i give-up. I'l compile the original testZigbee.cpp (without any changes like the date and checksum value. This must be completely identical to what you are sucessfully running.
[testZigbee_original](https://github.com/patience4711/RPI-APS-inverters/blob/main/testZigbee_original.cgi)
Please check the permissions like above

autourdupc, 2023-12-28:
Same as you...

[image]

testZigbee_original.cgi -> Works fine !
[image]

patience4711, 2023-12-28:
crazy, the only thing that is stil open; I compiled it on my 3_3 system. I copied the cpp file over from 3_3 to my working 3_9a and compiled it there. That is now testZigbee.cgi and it works as normal. 

I start wondering; could it be that files compiled on the 3_3 system work for you but when compiled on the 3_9 wil fail? I made the cpp file [here](https://github.com/patience4711/RPI-APS-inverters/blob/main/testZigbee.cpp)
maybe you can compile it on your 3_9 system to check if it stil works. It's up to you, my system works fine with all versions. But if you want to solve the riddle ...

autourdupc, 2023-12-28:
Took your file and named it testZigbeefrom3_3.cpp (hope it comes from 3_3)

Compiled on my 3_9a (from your SD card image)
It works !

[image]

autourdupc, 2023-12-28:
> var w is the data, returned by chartdata_pe.php

Sure !
I was blind during one week... I mixed chartdata_pe and chart_pe

patience4711, 2023-12-28:
The problem is in this function (old an new version below). 
The difference is that in this line **sprintf(bufferCRC, "%02x", StrToHex(bufferCRC) ^ StrToHex(bufferCRCdiezweite));**  %02**x** (old) became %02**X** (new uppercase) and the 
 **uint8_t iToUpper = 0;
     while (bufferCRC[iToUpper])
    {
        bufferCRC[iToUpper] = toupper(bufferCRC[iToUpper]);
        iToUpper++;
    }** 
was commented out in new (not needed anymore because the sprintf already took care for the uppercase)

### old: (working for you)
// *************  calculate  the checksum of the message ****************************
char *checkSum(char Command[])
{
    char* bufferCRC = new char[254];
    char bufferCRCdiezweite[254] = {0};

    strncpy(bufferCRC, Command, 2); //as starting
    delayMicroseconds(250);         //give memset

    for (uint8_t i = 1; i <= (strlen(Command) / 2 - 1); i++)
    {
        strncpy(bufferCRCdiezweite, Command + i * 2, 2);
        delayMicroseconds(250);
        sprintf(bufferCRC, "%02x", StrToHex(bufferCRC) ^ StrToHex(bufferCRCdiezweite));
        delayMicroseconds(250); //give memset a little bit of time to empty all the buffers
    }
    uint8_t iToUpper = 0;
     while (bufferCRC[iToUpper])
    {
        bufferCRC[iToUpper] = toupper(bufferCRC[iToUpper]);
        iToUpper++;
    }
    return bufferCRC;
}
### new: not working for you
// *************  calculate  the checksum of the message ****************************
char *checkSum(char Command[])
{
    char* bufferCRC = new char[254];
    char bufferCRCdiezweite[254] = {0};

    strncpy(bufferCRC, Command, 2); //as starting
    delayMicroseconds(250);         //give memset

    for (uint8_t i = 1; i <= (strlen(Command) / 2 - 1); i++)
    {
        strncpy(bufferCRCdiezweite, Command + i * 2, 2);
        delayMicroseconds(250);
        sprintf(bufferCRC, "%02X", StrToHex(bufferCRC) ^ StrToHex(bufferCRCdiezweite));
        delayMicroseconds(250); //give memset a little bit of time to empty all the buffers
    }
    //uint8_t iToUpper = 0;
    // while (bufferCRC[iToUpper])
    //{
    //    bufferCRC[iToUpper] = toupper(bufferCRC[iToUpper]);
    //    iToUpper++;
    //}
    return bufferCRC;
}
It is only a simple modification that affects the conversion to uppercase, not the output. I can't see why it would make a difference on your system. Maybe you can see it.
If you are going to experiment you could start with comment out the delayMicroseconds(250)'s,  i am not sure if they are needed. When i do that it works fine but your platform is faster.

autourdupc, 2023-12-29:
I start to do some tests...

I put in the same function both codes to calculate CRC and display the results... Both are the same.
I also removed the delays and it still working.

```
// *************  calculate  the checksum of the message ****************************
char *checkSum(char Command[])
{
    char* bufferCRC = new char[254];
    char bufferCRCdiezweite[254] = {0};

    char* bufferCRClbo = new char[254];
    char bufferCRCdiezweitelbo[254] = {0};

    strncpy(bufferCRClbo, Command, 2); //as starting
    delayMicroseconds(250);         //give memset

    for (uint8_t i = 1; i <= (strlen(Command) / 2 - 1); i++)
    {
        strncpy(bufferCRCdiezweitelbo, Command + i * 2, 2);
        //delayMicroseconds(250);
        sprintf(bufferCRClbo, "%02x", StrToHex(bufferCRClbo) ^ StrToHex(bufferCRCdiezweitelbo));
        //delayMicroseconds(250); //give memset a little bit of time to empty all the buffers
    }
    uint8_t iToUpper = 0;
     while (bufferCRClbo[iToUpper])
    {
        bufferCRClbo[iToUpper] = toupper(bufferCRClbo[iToUpper]);
        iToUpper++;
    }
    cout << "bufferCRC old method : "<< bufferCRClbo << "\n<br>" << endl;   
    //return bufferCRClbo;

    strncpy(bufferCRC, Command, 2); //as starting
    delayMicroseconds(250);         //give memset

    for (uint8_t i = 1; i <= (strlen(Command) / 2 - 1); i++)
    {
        strncpy(bufferCRCdiezweite, Command + i * 2, 2);
        //delayMicroseconds(250);
        sprintf(bufferCRC, "%02X", StrToHex(bufferCRC) ^ StrToHex(bufferCRCdiezweite));
        //delayMicroseconds(250); //give memset a little bit of time to empty all the buffers
    }
    cout << "bufferCRC new method : "<< bufferCRC << "\n<br>" << endl;   
    return bufferCRC;
}
```

[image]

So the issue is not on the "%02X" part...

I still continue to investigate.

[23 more replies]

closed 2024-01-12 by patience4711