patience4711/RPI-APS-inverters #99: Countdown to next polling cycle

autourdupc, 2023-12-13:
Hello.

Would it be a good idea to have a countdown displaying to the next polling cycle ?

patience4711, 2023-12-15:
Yes, but i need a strong argumentation for what this is usefull......

autourdupc, 2023-12-16:
Just to know when new values will come... (I know, this is not a strong argumentation :) )

Maybe also the signal strenght in front of each of inverters... This could be visible on the main page.
I think you got it when polling inverters... I could see "13:22:03 polled inv 1 ts: 16973 signal q: 48.2%" 

And why not the graph on the main page ?

I know,, I ask for too many things, but I find this project so marvellous... Maybe I can try to modify by myslef if it is not in a CGI file (My C programming is so far... I was young).

closed 2023-12-18 by patience4711