patience4711/RPI-APS-inverters #101: YC600 with 1 solar pannel : No buttons for graphics

autourdupc, 2023-12-14:
Hello.

Got an YC600 connected only with 1 solar pannel
There is no graphic button available nor statistics button.
[image]

Got a QS1 with 4 solar pannels and got thoses buttons.
[image]

patience4711, 2023-12-14:
This seems impossible to me, i just tried to reproduce it with my own yc600 and get this result. This looks perfectly oke. The page displays the buttons whenever the panelcount > 0. So it seems to think that panelcount == 0.  What happened here? 
Maybe the inverterproperties file got corrupted somehow, can you examen /var/www/ecu-data/inverters/inv_properties<inverternr> should contain   a 1 for each panel like: "panels":[0, 1, 0, 0]  

If not you can edit it. Or edit the inverterpage and save it. I should be allright then.

If the properties are oke you could try to refresh the page and/or empty the browsercache.

[image]

autourdupc, 2023-12-14:
Here is the file content when opened ...
{"id":"0x2C1D","serial":40xxxxxxxxx3,"type":0,"name":"YC600-1","idx":"2","panels":[0, 1, 0, 0]}

And inverter page when opened
[image]

After refreshing, got same result.
I also changed browser... Same issue.

autourdupc, 2023-12-14:
Got the same issue with QS1 if I only check panel #2

[image]

[image]

patience4711, 2023-12-14:
This problem is very hard to solve if i cannot reproduce it. 

I have published a new software upgrade to v3_9.  In order to test, I just installed a fresh image v3_3 on a brand new zero W and upgraded to v3_9. 
Now with firefox running on a linux system, I tried  all kinds of things to fuck up with the inverterpage but it always works as expected. Always when there is at least one panel checked, no matter which one,  the buttons are displayed.
I suggest that you also upgrade to 3_9.

If you still have this problem then you can debug by editing the file /var/www/html/invpage.php.
Find the line **if(panelcount > 0)** and just before this line you add **echo "panelcount = " . $panelcount;**
Also find the line **//echo "panels". $p0a . $p1a. $p2a . $p3a;** and remove the slashes.
Save and visit/refresh the page. Now you can see which panels are checked and what the value of $panelcount is. If $panelcount > 0 the buttons should be displayed. 

If not, check the dir /var/www/ecu-data and make sure that there are files for each inverter (and no more) and check for unusual things.

As for the name mismatch, i don't know what you mean by that. This can only occur when the properties files are corrupted.

autourdupc, 2023-12-14:
I will do the tests you asked and let you know. Need for that some time. Tomorrow I'm busy. Hope to come back not too late to do some.
I will update to 3_9 and let you know.
Anyway, I can also give you a TeamViewer session on my computer so you can access my browser and my RPI-ECU if needed.

autourdupc, 2023-12-14:
@patience4711 
Would like to share a video... How to MP directly to you not public ?

patience4711, 2023-12-15:
That's another mystery, and maybe that is the whole problem. After i updated, the version 3_9 is displayed correctly. In the update script there is a line that updates the file /var/www/ecu_data/swVersion.txt. While the update is installed, all is logged so you can exactly see what is happening, you should be able to see if there are errors. So you can conduct the update again without problems and examen the output.

I have the feeling that there is something messed up with the file permissions, otherwise i couldn't explain why this doesn't work. Check if the file /var/www/ecu_data/swVersion.txt (and the whole directory) is writeable by owner and that the owner is www-data.  
Or maybe your sdcard is corrupt. You use an 8Gb card probably an old one?

Look, i really want to help you but this costs me too much time. To me (and many others) the software works as it should so you must be doing something wrong. 
Dont you have an old raspberry B that you can use to test with? To rule out that your hardware is the cause.
Or start all over with a new class10 sdcard with version 3_3 without updating and check if it works.

patience4711, 2023-12-15:
I made an image of the working version v3_9 and published that. I am going through the whole proces of downlaoding, burning the sd an install, just to be sure that it all works oke.

autourdupc, 2023-12-17:
@patience4711 
As I said in another post, all is working fine except panels swapped only for QS1.

I did some hack of the php pages for fun...
And because I was not able to put the chart in the table, I used an iframe with some parameters passed to "chart_pe.php"
[image]

This is only working on computers, phone devices need to use landscape mode to get a correct page due to size and not responsive code.

patience4711, 2023-12-18:
Really nice. Although when you have 9 inverters....

autourdupc, 2023-12-18:
Hum... I formatted the SD card and put the new image... No trace of what previously was wrong.
I notice that the html is not "clean" enought... Missing some closing instructions (like </td> or </tr>)
Also some deprecated instructions and maybe globaly some issues with styles... 
Maybe bad headers also... Not yet checked.
All of that could create unwanted results.
And also, its hard to follow the code due to php/html mixed so many times (ah ah ah !)

I also did poor html quality for testing multiple charts on the same page - It needs some time to read, check, read again, check again...

closed 2023-12-22 by patience4711