patience4711/RPI-APS-inverters #105: Questions about updates

autourdupc, 2024-01-22:
I would like to get details concerning updates...

1) If I put some extra files in folders for personnal use (php files, cgi files, javascripts), or even if i install extra packages, what happens if I update to next version ?

2) Does the systems needs updates via apt ?

Thansk for details.
Regards,

patience4711, 2024-01-24:
@autourdupc  The only thing that will happen is that the directories "/var/www/html" and "/usr/lib/cgi-bin/ecu" are completely removed and re-installed. 
This the most convenient for me because when i would have to make a script to replace only the nessecary files (like i did in the beginning) it would give me a lot of work, administration and chances of errors. It is also simple way to remove redundant files.

So you only have to keep your own files outside these directories and youl'l be fine. But the document root of apache2 is /var/www/html so you probably have to change that. Or maybe keep your files in a subdirectory that you backup before a software update and place back afterwards.

The distro updates are not needed as far as i know but it will not hurt.

autourdupc, 2024-01-25:
@patience4711 
Thanks for replying.
I will consider, if I find time to "play" with extra scripts, to backup or duplicate to another folder before updating.

Concerning distro updates, I like maintening the system to the last version, for security, enhancement, and sometimes optimizations.

closed 2024-02-03 by patience4711