patience4711/RPI-APS-inverters #107: Uploaded photo not showed - Deleted photo not deleted (RPI-ECU-v3_9)

autourdupc, 2024-02-16:
Hello.

Using RPI-ECU-v3_9

I'm trying to upload a photo.
Tested on Firefox and Edge... Not working.
Strangely, on Firefox, a pop-up asks for a new upload. Whatever your choice is, it is not good. See below.
The photo is not uploaded while the message says its okay.

When trying to deleted the exiting photo (your photo), its says its okay, but not. Your photo remains.

Here is an upload sequence...
[image]

[image]

[image]

[image]

[image]

Best regards,

patience4711, 2024-02-21:
@autourdupc  did you see the help button? Useually emtying the browsercache will do the trick,

autourdupc, 2024-02-22:
@patience4711 
You are the king.
Using a second computer provide the right photo after some times...

closed 2024-02-22 by autourdupc