patience4711/RPI-APS-inverters #108: Error 100

jmguilbert, 2024-02-21:
Good morning
In the morning, my rpiecu works very well. The "pwr" and "en" come back well in the application and by Mqtt. 
But after a few hours, some inverters no longer go up.
And in the logs there are:
**15:55:10 polled inv 3 error 100 invalid energy value
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData3** 

Do you have any ideas for work or research?
THANKS.

jmguilbert, 2024-02-22:
Hey goomorning
In the log, i found this :+1: 10:45:25 polled inv 5 ts: 9435 signal q: 45.1%
10:50:08 polled inv 1 ts: 290 signal q: 43.1%
10:50:10 polled inv 2 ts: 290 signal q: 40.0%
10:50:12 polled inv 3 ts: 290 signal q: 17.3%
Can you indicate me what is it ?
Thanks

patience4711, 2024-02-23:
The redcolor means that the timespan is not 5 minutes like it should. If an inverter wokes and is polled, and then goes down, and later goes up again, the timespan between the polls is 8244 seconds. In your case it seems that the inverters woke up very late or go down too soon. Probably due to a lack of daylight.

jmguilbert, 2024-02-24:
Thank you Patience4711.

closed 2024-03-11 by patience4711